openhwfoundation/cv32e40p #119: issue with DPC register behavior (same is with MEPC)

Itaykeidar, 2019-08-21:
Here is definition of this register from the specification:
[image]

Test executed following instruction: csrrsi  a6, dpc, 27
27 in this print is decimal value. As the result of instruction execution register a6 will contain old value of dpc register and
dpc register will be updated by next formula: dpc = dpc | 27.
Old value of dpc register is 0xa. Following upper formula new value should be dpc = 0xa | 0x1b = 1b.

In this case spike calculated it differently. DPC register in the spike is 0x1a.
I tried to debug this issue and I found that spike performs some alignment of dpc register value.

[image]

Path to source code on github: https://github.com/riscv/riscv-isa-sim/blob/master/riscv/processor.cc

Can you please advise if this is a bug / spec clarification, or do we need for such address alignments of dpc register on spike side 

as stated in title, this is in mepc reg same behavior.

bluewww, 2019-08-21:
Thanks for the report. It looks like what spike does makes sense: The specification doesn't allow the instructions to be aligned with any other value than 16 or 32 bits, which also implies constraints on the program counter. I suggest we enforce this constraint too.

bluewww, 2019-08-21:
See #120

yaronbe1, 2019-09-18:
Hi

as stated in the bug description, same bug is present also in MEPC register.

the fix AFAIK only fixes the DCP reg.

yaronbe1, 2019-09-19:
also, the fix is done only to PULP_SECURE==1 section.

should also be applied to  PULP_SECURE==0 section (line 775)

please review your fix

bluewww, 2019-09-19:
Thanks I'll look into it

bluewww, 2019-09-19:
The `dpc` fix is applied to both `PULP_SECURE` modes, though you are right that `mepc` is not yet changed.

PR #163 (2019-09-19): Fix alignment of mepc
riscv-privileged-v1.10, p. 34: The low bit of mepc (mepc[0]) is always
zero.
Fixes #119
closed 2019-09-19 by bluewww