openhwfoundation/cv32e40p #126: Combinational paths between RVALID_I and REQ_O

arjanbink, 2019-08-26:
Combinational paths exist from data_rvalid_i to data_req_o and from instr_rvalid_i to instr_req_o. Such paths can potentially limit the achievable system performance.

See attached pdf for details.

[RI5CY_Bus_Interface_0.pdf](https://github.com/pulp-platform/riscv/files/3541749/RI5CY_Bus_Interface_0.pdf)

bluewww, 2019-08-26:
Did you try to run the tests in `tb/core` with your proposed changes?

arjanbink, 2019-08-26:
I did not. If these test would pass I would still not consider that a 'good enough proof' though. Please also note that I only proposed a potential fix to the load/store unit; the ifetch code is more complicated and I did not look at that (other than observing a combinational path exists

davideschiavone, 2019-08-29:
Just as a reference: https://github.com/lowRISC/ibex/issues/265

davideschiavone, 2019-09-02:
Ibex possible solution: https://github.com/lowRISC/ibex/pull/276

closed 2020-06-15 by Silabs-ArjanB