openhwfoundation/cv32e40p #129: Non-reachable FSM states in load/store unit

arjanbink, 2019-08-26:
The WAIT_RVALID_EX_STALL and   IDLE_EX_STALL states in the load/store unit FSM are not reachable.

**Git tag:** 

eae1cabcdb2333e5f1037996a48504985e7813a7 (august 5   2019)

**Parameters:**

(PULP_SECURE, USE_PMP, PULP_CLUSTER)   = (0, 0, 0) or (1, 1, 0) or (0, 0, 1) or (1, 1, 1) 
N_EXT_PERF_COUNTERS = 0   
INSTR_RDATA_WIDTH = 32   
N_PMP_ENTRIES = 16   
FPU = 0   
Zfinx = 0   
FP_DIVSQRT = 0   
SHARED_FP = 0   
SHARED_DSP_MULT = 0  
SHARED_INT_DIV = 0   
SHARED_FP_DIVSQRT= 0  
WAPUTYPE = 0   
APU_NARGS_CPU = 3   
APU_WOP_CPU = 6   
APU_NDSFLAGS_CPU   = 15   
APU_NUSFLAGS_CPU   = 5   
DM_HaltAddress = 32'h08000000

The WAIT_RVALID_EX_STALL and IDLE_EX_STALL states of CS (current state) are not reachable for the above listed parameter configurations. Is this dead/obsolete code in the RTL, or are these states reachable for other parameter configurations, or are these states intended for future extensions to RI5CY?

davideschiavone, 2019-08-29:
We also noticed it at : https://github.com/pulp-platform/riscv/issues/80

We are investigating

davideschiavone, 2019-12-11:
removed in new_pipeline branch. Under evaluation

3bfe1ea820e60e3239a3d9b3c8ef403c45d4c3c1

closed 2020-06-15 by Silabs-ArjanB