openhwfoundation/cv32e40p #131: Addition of 'not sleeping pin'

arjanbink, 2019-08-27:
The environment of RI5CY cannot judge whether RI5CY still requires its clock (or whether it is for example blocked in a WFI). Addition of a ‘not sleeping’ pin can provide this information. At chip level this information can be used for power management. This feature request amounts to simply making the already existing clock_en signal in riscv_core.sv available on a newly added cpu_sleeping_n pin (see attached pdf).

[RI5CY_Not_Sleeping_Pin.pdf](https://github.com/pulp-platform/riscv/files/3544751/RI5CY_Not_Sleeping_Pin.pdf)

Note: This feature has also been requested for Ibex (https://github.com/lowRISC/ibex/issues/258)

davideschiavone, 2019-08-28:
Probably you can use core_busy_o.
1 if the core is doing something, 0 otherwise.

arjanbink, 2019-08-30:
core_busy_o does not provide enough information to judge whether RI5CY needs a clock or not. Suppose we would use (only) core_busy_o to gate clk_i, then there would be no way to wake up from WFI anymore (as there is no combinational path from irq_i to core_busy_o; the same holds for debug_req_i). You would really need to have external access to (or externally duplicate)  RI5CY's clock_en signal. (Alternatively the semantics of core_busy_o could be changed, but of course other people/circuits might depend on its current definition, so it is probably not a good idea to change that).

Silabs-ArjanB, 2020-07-08:
Fixed with https://github.com/openhwgroup/cv32e40p/pull/391

closed 2020-07-08 by Silabs-ArjanB