openhwfoundation/cv32e40p #132: MPP of MSTATUS CSR register written with the wrong privilege level

shetalani, 2019-08-27:
Here is an issue found after applying OneSpin 360 DV-Verify RVV App on RI5CY:

**Description:** we have a load to an address where reads are not allowed followed by a fetch from an address that is outside the instruction match region. As the latter is already taken in decode, it happens before the access fault of the load gets served. However, it gets overwritten by the load access fault before the next instruction gets executed - which almost makes it good again. But one thing that goes wrong is the storing of the privilege level where the exception occurred. The false fetch access error sets the privilege to M mode; thus, the preceding load access fault thinks it was executed in M mode while it was in U mode.
**Example:** for a set of instructions: 32’h3a0a5e73 (csrrwi x28,pmpcfg0,20), 32’h200073 (uret), 32’h1941413 (slli x8,x8,25), 32’h12083 (lw x1, 0(x2)) being decoded at the following time points t##-7, t##-6, t##-1 and t##0, with the following pc values in the decode stage 32’h24, 32’h28, 32’h0 and 32’h2 respectively. Knowing that these instructions were fetched from the memory at the timepoints t##-9, t##-8 and t##-3 respectively (The last two instructions are compressed). Fetching an instruction from the address 32’h4 causes an IAF since this address is outside the instr match region specified by the pmpcfg0 register. This IAF is handled at time point t##2, so that the current privilege level moves back from U to M mode and the CSRs mepc, mcause and mstatus are written with the associated values. However, the LW instruction tries to read the memory at time point t##1 when the privilege level is U mode still and since R bit is cleared in pmpcfg0, this causes a LAF that is handled at time point t##4. This time, mepc, mcause are updated correctly, but mstatus is not. MPP field is written with the value 2’h3 which refers wrongly to M mode, as it should be the value associated with U mode since the LAF happens in the U mode.

**The same issue could be encountered in case we move to debug mode while executing a load instruction that causes a LAF in U-mode.**

[image]

------------------------------------------------------

Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2019.2.1

shetalani, 2019-11-14:
In general, if there is an instruction being decoded/executed and one of the following scenarios happens:

1. an illegal instruction exception caused by the next fetched instruction
2. ecall or ebreak instruction fetched next
3. a fetch access exception caused by the next fetched instruction, clarified in this issue
4. debug mode is entered during the execution of the instruction:
     1. due to a halt request, reported in this issue
     2. due to an ebreak with forced entry

In all these cases, the exceptions update CSRs before the instruction being decoded, in case it causes an exception itself, does. 

Noting that for case 4. the debug mode is entered without updating dcsr, mstatus and mcause are updated instead, reported in #157. Also, the exception caused by the instruction being decoded/executed would be handled in debug mode, which is another example of #185.

silabs-PaulZ, 2020-06-10:
Hi @shetalani , @davideschiavone ,

With the latest RTL, scenarios 1, 2, & 4 are resolved. 

Scenario 3 is no longer valid as this requires PMP which is not supported on CV32E40P. Note, this may be an issue to look at on future cores.

Below are some examples showing various scenarios behaving correctly. Please, if you can, rerun your formal tool/simulation to confirm as well. If you agree with the fix, **please close this issue.** If we do not get a response within a week then we will automatically close this issue.

Scenario 2) ebreak call (non debug entry). The last instruction is executed before MSTAUTS, MCAUSE, MEPC are updated and the core jumps to ebreak exception routine. The series of instructions are adding to the a5 register, where the last add executes before going through the FLUSH_EX and FLUSH_WB states.
 ```
    bca2:	0795                	addi	a5,a5,5
    bca4:	0799                	addi	a5,a5,6
    bca6:	079d                	addi	a5,a5,7
    bca8:	9002                	ebreak

```
 
[image]

Scenario 4a) debug entry due to debug request. The last instruction is executed before DEPC, DCSR registers are updated and the core jumps to debug mode execution. In particular, the state machine enters DEBUG_FLUSH to ensure a clean transition. 
```
    bd2a:	  07a1                	addi	a5,a5,8
    bd2c:	  07a5                	addi	a5,a5,9
    bd2e:	  07a9                	addi	a5,a5,10 <- finish execution before DCSR,DEPC is updated
   < **debug_req_i asserted**>
    bd30:	  07ad                	addi	a5,a5,11 <- debug_req prevents execution
    bd32:	  07b1                	addi	a5,a5,12
```
[image]
 

Scenario 4b) debug entry due to ebreak. Similar to debug_req scenario above
```
    be66:	0789                	addi	a5,a5,2
    be68:	078d                	addi	a5,a5,3
    be6a:	0791                	addi	a5,a5,4
    be6c:	0795                	addi	a5,a5,5
    be6e:	9002                	ebreak
    be70:	0799                	addi	a5,a5,6
```
[image]