openhwfoundation/cv32e40p #137: Violation - Compressed instruction is not valid

shetalani, 2019-08-29:
Here is an issue found after applying OneSpin 360 DV-Verify RVV App on RI5CY:

**Description:** While "The RISC-V Instruction Set Manual Volume I: Unprivileged ISA" states that: "C.JR is only valid when rs1̸=x0; the code point with rs1=x0 is reserved.", the core doesn't raise an illegal instruction exception for such a case.
**Example:** The instruction 32'h8002 (c.jr x0) is decoded at time point t##0 with no illegal instruction being flagged, as illegal_insn_dec is de-asserted. 

[image]

--------------------------------------------
Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2019.2.1

Silabs-ArjanB, 2020-06-02:
Hi @shetalani This issue has now been fixed. Could you please close this issue if you agree it has been solved. The issue will automatically be closed on June 10 2020 otherwise.

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.

closed 2020-07-14 by Silabs-ArjanB