openhwfoundation/cv32e40p #141: Verilator error on fpnew_divsqrt_multi.sv

aignacio, 2019-09-04:
Hi pulp-team,
I'm facing some issues due to compilation of verilator model with the fp hardware division enabled and fpu. The errors reported from verilator are related to the dual-type assignments made on the file **fpnew_divsqrt_multi.sv** in the lines 296/297/298/299/300, as can be seen through the image below. Is there some different file to include once you want to generate the verilator model with the parameters FP_DIVSQRT=1?
PS.: This error just happen once you enable the hw div. unit, if you enable just the FPU, the model can be generated without any errors.

[image]

bluewww, 2019-09-04:
Thank you for your report. Verilator unfortunately doesn't fully support SystemVerilog so unfortunately breakage like this can happen.

Infact there is a similar issue if you enable `PULP_SECURE=`1 and try to run the verilator testbench. I can try to fix the latter.

stmach, 2019-09-04:
You'll have to disable the strict BLKANDNBLK checking for Verilator by using the -Wno-BLKANDNBLK option.
There might be some other ones as well, you can refer to the list of options we use on Ariane here: https://github.com/pulp-platform/ariane/blob/0cb44658501b1a4b29c567c002ab2b7b779c190a/Makefile#L349

aignacio, 2019-09-04:
you're right @stmach, now it compiles without the problems!

closed 2019-09-04 by aignacio