openhwfoundation/cv32e40p #142: Change the address of some custom CSRs to be in the range reserved for such CSRs

arjanbink, 2019-09-04:
Some of RI5CY's custom CSR are in non-custom ranges and these might therefore cause conflicts with future standard extensions. Also the Hardware Loop CSRs are in the Machine CSR section, which would become an issue once https://github.com/pulp-platform/riscv/issues/130 has been resolved.

My proposal would be:

- Move HWLoop0_START, HWLoop0_END, HWLoop0_COUNTER, HWLoop1_START, HWLoop1_END, HWLoop1_COUNTER somewhere in User CSR custom read/write space (0x800-0x8FF)

- Move duplicated mhartid from 12'h014 into User CSR custom read-only section (0xCC0-0xCFF)

- Move current priv level from 12'hC10 into User CSR custom read-only section (0xCC0-0xCFF) (as 12'hC10 is reserved for hpmcounter16).

bluewww, 2019-09-04:
Are we allowed to just duplicate the hwloop csrs into the user csr custom read/write space?

arjanbink, 2019-09-04:
I am not 100% sure about the spec in this respect. I believe that if you put the HW Loop CSRs in the user csr custom part, then you can use them in both user and machine mode (but if you leave them where they are now, then you should not be able to access them in user mode). I assume that you intend these CSRs to be available at the lowest privilege level, right? I do not see the point in duplicating these registers.

bluewww, 2019-09-04:
Indeed it makes sense to have these registers in user mode. It is just a bit painful to remap CSRs because it will break our SDK. I'll have a look with @haugoug .

bluewww, 2019-09-05:
I made a table overview of the mappings

MarekPikula, 2019-09-25:
In addition I can see that performance counters don't use the standard addresses. Look in [Privileged spec](https://github.com/riscv/riscv-isa-manual/releases/download/draft-20190913-682271d/riscv-privileged.pdf) in table 2.5 and from 3.1.11 section. It makes SDK incompatible with Ibex for PULPissimo and with spec, what I guess is more important.

[Here](https://ibex-core.readthedocs.io/en/latest/performance_counters.html) are docs of Ibex implementation.

MarekPikula, 2019-09-25:
And here is the PR, which changed performance counters in Ibex: lowRISC/ibex#53

Silabs-ArjanB, 2020-06-19:
Above pull request fixes various CSR addresses as follows:

```

CSR                     Old address     New address     Comment
---------------------------------------------------------------
HWLoop0_START           12'h7C0         12'h800         Needs to be User accessible (now in user custom read/write space)
HWLoop0_END             12'h7C1         12'h801         Needs to be User accessible (now in user custom read/write space)
HWLoop0_COUNTER         12'h7C2         12'h802         Needs to be User accessible (now in user custom read/write space)
HWLoop1_START           12'h7C4         12'h804         Needs to be User accessible (now in user custom read/write space)
HWLoop1_END             12'h7C5         12'h805         Needs to be User accessible (now in user custom read/write space)
HWLoop1_COUNTER         12'h7C6         12'h806         Needs to be User accessible (now in user custom read/write space)
FPREC                   12'h006         12'h807         Cannot be in Standard space (now in user custom read/write space)
UHARTID                 12'h014         12'hCC0         Cannot be in Standard space; should not be R/W (now in user custom read-only space)
PRIVLV                  12'hC10         12'hCC1         Cannot be in Standard space (also clashes) (now in user custom read-only space)

```

davideschiavone, 2020-09-30:
shall we close this issue? @Silabs-ArjanB

closed 2020-09-30 by Silabs-ArjanB