openhwfoundation/cv32e40p #144: dpc increment based on cause

dspg-arch1, 2019-09-04:
there is a question that arises on how is the SW supposed to handle the different dpc increments according to the break reason.

From simulation   we understand, the dpc register modification is
following by these rules:
- increase dpc by 2 if debug session is initiated by c.ebreak
- increase dpc by 4 if debug session is initiated by ebreak
- no changes of dpc if debug session is initiated by debug request signal

Is this correct? if so, how we can determine from SW side the reason of entering into the debug mode?
how is SW supposed to handle debug ROM code?

bluewww, 2019-09-04:
The `dpc` changes are listed in the `riscv-debug-release` v0.13.2 document on page 44.

The specification is called `RISC-V External Debug Support`, because it is meant to be used, well, externally (over JTAG or whatever) to the core. I imagine you could use a memory mapped interface (instead of the JTAG DMI) to debug one core with another.

If you want to use software debugging (to debug a core itself) then you would use only parts of the debug module such as hardware triggers and hardware/software breakpoints. You then can control where the control-flow goes to by setting/unsetting `ebreakm`, `ebreaku` and `ebreaks`. If set you jump into the debug module (now only extenral debugging is possible), if unset you have the behaviour available that is defined the privilege spec. Typically this would mean jumping to your software debug routine.

silabs-PaulZ, 2020-05-19:
I agree with @bluewww and would like to expand their answer. Below are some scenarios on how to handle debug and ebreak. If all agreed, then @dspg-arch1 please close the issue. If no response, we should close this issue by  5/25/2020.

- **ebreak call entry into software exception handler**
  - Assumptions:
    - debug_mode == 0
    - dcsr.ebreakm == 0
  - Result in the following:
    - Entry into exception handler routine located at mtvec, debug mode not entered
    - mepc = address of ebreak or c.ebreak
    - mcause = ebreak exception
    - dpc, dcsr.cause, debug_mode are not modified
  - SW action (in exception handler):
    - SW must move mepc to next instruction before returning from exception routine.
        - Requires reading instruction located at mepc to determine the size of instruction (16 bit c.ebreak or 32 bit ebreak based on instr[1:0] ) and then increment mepc (with 2 or 4)
        - CV32E40P does not support mtval. Mtval, if implemented, would capture the ‘bad’ illegal instruction, removing the need to query the instruction from memory. This may be supported in future cores.

- **ebreak call entry into debug mode**
  - Assumptions
    - debug_mode == 0
    - dcsr.ebreakm == 1 
  - Results in the following: 
    - Entry into debug mode and execution of debug code at DM_HALTADDRESS
    - debug_mode = 1
    - dpc = address of break
    - dcsr.cause = ebreak (0x1)
  - SW action (in debugger code):
    - SW must move dpc to next instruction before returning from debugger exit (dret)
      - Requires reading instruction located at dpc to determine the size of instruction (16 bit c.ebreak or 32 bit ebreak based on instr[1:0] ) and then increment mepc (with 2 or 4)

- **ebreak call to restart debugger code**
  - Assumptions
    - debug_mode == 1
    - dcsr.ebreakm == NA
  - Results in the following: 
    - jump to beginning of debug code at DM_HALTADDRESS
    - status are not modified (dpc, dcsr.cause not modified)
  - SW action: NA
    - dpc adjustment depends on debug entry cause, not from executing ebreak while in debug mode.

- **External debug request (debug_req) into debug mode**
  - Assumptions
    - debug_mode == 0
    - dcsr.ebreakm == NA
  - Results in the following:
    - Entry into debug mode and execution of debug code at DM_HALTADDRESS
    - debug_mode = 1
    - dpc = address of next instruction to be executed
    - dcsr.cause = haltreq (0x3)
  - SW action (in debugger code): 
    - Do not adjust dpc. Simply call dret when done.

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it has been answered at least 7 days ago. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.

closed 2020-07-14 by Silabs-ArjanB