openhwfoundation/cv32e40p #148: Instruction corrupted on return from interrupt to last instruciton of HW loop

dspg-arch1, 2019-09-05:
Issue happen when ri5cy tried to execute instruction with PC 0x800002fa. Instruction is 0x03f699b3. Decoded instruction is "mulh s3,a3,t6".
This instruction is the last instruction of the HW loop #0. We can see in the waveform signal "hwlp_end[0]" is set to 0x800002fa.
We can also conclude that IRQ interrupted execution of this instruction because MEPC register set to 0x800002fa.

We expect to see after execution of mret instruction (end of interrupt handling procedure), execution of instruction with PC 0x800002fa and jump to the start of the HW loop #0.
In the waveform we see that instruction 0x8C0199B3 was executed instead of 0x03f699b3 as it described in the firmware.
Instruction 0x8C0199B3 was decoded as illegal instruction by ri5cy.

Here is path of the firmware with lines describing loop #0:

80000264: 8c01 sub s0,s0,s0
...
800002fa: 03f699b3 mulh s3,a3,t6
From this lines we can conclude that ilegal instruction 0x8C0199B3 that ri5cy tried to execute is concatenation of 2 instructions located at the beginning of the loop and at the end of the loop.
[image]

dspg-arch1, 2019-09-05:
Can be disregarded, duplicated issue, see 107

dspg-arch1, 2019-09-16:
might be separate bug
We are not sure if this is a separate issue/bug please check are merge is they are the same

davideschiavone, 2019-09-19:
Can you please check again if this test passes with this fix?
https://github.com/pulp-platform/riscv/commit/78c4c448bdc7cb0094c2d759ca5352236e7e44c6

yaronbe1, 2019-10-02:
after applying the fix we have found the following issue:

[image]

Here we can see active HW loop #1 with start address 0x80000424 and end address 0x80000582. 
Disassembled instruction with address 0x80000424 is "80000424: 694055d7           pv.and.sc.b a1,zero,s4".
Disassembled instruction with address 0x80000582 is "80000582: 1041cf57           pv.avg.sc.h t5,gp,tp".
On execution of the last instruction of HW loop body IRQ occurred (see the first timepoint). 
After leaving IRQ processing function (execution of mret instruction) we can see illegal instruction detected with PC 0x80000582.

If we analyze instruction that ri5cy tried to execute with PC 0x80000582,  we can see that it differs from instruction in the firmware and composed from parts of instruction at the beginning and end of the loop body.

Data on the firmware :
80000424: 694055d7
80000582: 1041cf57
Executed instruction
80000582: 55D7CF57

for DSPG internal: To open simulation results use next command:
lbrun -C simvision -p high -- simvision -64bit -waves /home/dtmp/run/dbm10l/denisg/ri5cy/ws_dbm10l_test/prefetch_buffer_glitch/vlg_sim/waves.shm

davideschiavone, 2019-10-07:
On top of all the other changes I made for this bug, can you also add this one:

https://github.com/pulp-platform/riscv/commit/f19f0722b06b53a59cb38ea982d909f2d8797d13

please?

thanks
Davide

yaronbe1, 2019-10-09:
fix seems OK
you can commit

davideschiavone, 2019-10-10:
Thanks a lot, amazing news! does it also fix #158 and #107 ?

davideschiavone, 2019-10-10:
Ok then I wait for #158 before merging to master

davideschiavone, 2019-10-11:
Fixed by https://github.com/pulp-platform/riscv/commit/f19f0722b06b53a59cb38ea982d909f2d8797d13

davideschiavone, 2019-11-05:
The solution proposed contains a combinational loop,
please find the proposed fix for that combinatorial loop here:
https://github.com/pulp-platform/riscv/issues/190

I will merge to master once I get feedback from your functional verification and sorry for inconvenience

davideschiavone, 2019-11-12:
Is also this one fixed? @yaronbe1

yaronbe1, 2019-11-12:
> External Email - Use caution when responding, clicking and/or downloading attachments.

Yes

DSP Group, Inc. automatically scans all emails and attachments using MessageLabs Email Security System.

DSP Group, Inc. automatically scans all emails and attachments using MessageLabs Email Security System.

closed 2019-11-12 by davideschiavone