openhwfoundation/cv32e40p #149: complex arithmetic logic causes wrong results for non-complex sdotp operations

renzoandri, 2019-09-05:
I don't use the complex arithmetics, but it seems that with the sum-dot-product, the inversion is wrong, or actually it is inversing even though it is not a complex operation!
In the multiplier (line 283), for me it should be
<b>assign dot_short_op_a_1_neg = dot_short_op_a[1] ^ {17{(clpx_img_i)}};</b>
but it is: assign dot_short_op_a_1_neg = dot_short_op_a[1] ^ {17{(~clpx_img_i)}};
I've fixed it in my fork of the riscv core.

davideschiavone, 2019-09-11:
bfad7c12e5771e41e6e8256da5079a503a51d411

davideschiavone, 2019-09-11:
6d7d8d667e0b1a4fe9bb12e76b6ab29bd40c39d7
this commit solves the HW problem.

For past tapeout chips, the compiler should change the dot product encoding forcing the bit 25 to 1.

I keep the issue open as the immediate field of the dot product cannot use the bit 25 anymore.
It should be changed in tests, RTL and compiler

davideschiavone, 2019-12-20:
61593531b7cbb22549e3d18d704b509c450ec412

closed 2019-12-20 by davideschiavone