openhwfoundation/cv32e40p #152: Violation - Invalid instruction decoding

shetalani, 2019-09-06:
As it is in the specification, the xRET, ECALL, EBREAK, WFI instructions have the following encoding: 

[image]

[image]

However, RI5CY considers only the bits [31:20], [14:12] and [6:0] to decode such instructions and ignore the other ones. 

     OPCODE_SYSTEM: begin
        if (instr_rdata_i[14:12] == 3'b000)
        begin
          // non CSR related SYSTEM instructions
          unique case (instr_rdata_i[31:20])
            12'h000:  // ECALL
            begin
              // environment (system) call
              ecall_insn_o  = 1'b1;
            end

            12'h001:  // ebreak
            begin
              // debugger trap
              ebrk_insn_o = 1'b1;
            end

            12'h302:  // mret
            begin
              illegal_insn_o = (PULP_SECURE) ? current_priv_lvl_i != PRIV_LVL_M : 1'b0;
              mret_insn_o    = ~illegal_insn_o;
              mret_dec_o     = 1'b1;
            end

            12'h002:  // uret
            begin
              uret_insn_o   = (PULP_SECURE) ? 1'b1 : 1'b0;
              uret_dec_o     = 1'b1;
            end

            12'h7b2:  // dret
            begin
              illegal_insn_o = (PULP_SECURE) ? current_priv_lvl_i != PRIV_LVL_M : 1'b0;
              dret_insn_o    = ~illegal_insn_o;
              dret_dec_o     = 1'b1;
            end

            12'h105:  // wfi
            begin
              // flush pipeline
              pipe_flush_o = 1'b1;
            end

            default:
            begin
              illegal_insn_o = 1'b1;
            end

------------------------------------------------

Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2019.2.1

davideschiavone, 2019-09-19:
Can you please try this fix?
4071da186d690602033cae2aa86b081003e81765

thanks

shetalani, 2019-09-27:
It's fine now. Cheers!

davideschiavone, 2019-09-27:
Solved in 4071da186d690602033cae2aa86b081003e81765

closed 2019-09-27 by davideschiavone