openhwfoundation/cv32e40p #156: p.extract mismatch

yaronbe1, 2019-09-08:
we got a possible miscalculation in the p.extract instruction:

Source code of case:
                  li x2, 0x0
                  li x3, 0xFFFFFFFF
 
                  p.extract x2, x3, 2, 18
 
SPIKE Error message:
UVM_ERROR /proj/dbm10l/denisg/ws_dbm10l_initial/blk/ri5cy/verif/classes/pulpino_scoreboard_c.sv(249) @ 3553000: uvm_test_top.pp_env.pp_scoreboard [SPIKE_DBG] EX stage. Register #2 aka sp update mismatch. Expected value is 0x3, actual value is 0xffffffff
 
By definition from the spec:
p.extract	rD, rs1, Is3, Is2	rD = Sext((rs1 & ((1 << Is3) – 1) << Is2) >> Is2)
Note: Is3 + Is2 must be <= 32

  
Predicted manual calculation result following the formula should be: x3 = 0x3
ri5cy during simulation sets value of the register x2 =  0xffffffff.
this is inconsistent with manual calculation of the spec definition and with SPIKE.

RTL bug or spec wrong definition?

davideschiavone, 2019-09-19:
Hi @yaronbe1 ,

Sext is the sign extension of the extracted word "11".

Does it make sense for you?

yaronbe1, 2019-09-19:
as I understand from the formula, the 11  is logically shifted 18 bits to the left and than 18 bits back to the right. this gives the positive number of 00000000000000000011.
the Sext is done on this number, extending it's zero sign and you end up with 11 padded with 30 zeros from the left which is practically 0x3 .

davideschiavone, 2019-09-19:
no the sign extension is done on the extracted data, 11.

this means: take 2 bits from position 18 and sign extend it to 32 bits.

Best
Davide

davideschiavone, 2019-12-20:
61593531b7cbb22549e3d18d704b509c450ec412

closed 2019-12-20 by davideschiavone