openhwfoundation/cv32e40p #158: last instruction of the HW loop glitch happen without IRQs

yaronbe1, 2019-09-11:
I am opening this in a separate issue as this could be a stand alone possible bug, but it could be same issue as "IRQ arrive on the last instruction of HW loop #107". please merge if this is same cause.

relating to issue #107, we have managed to reproduced similar problem even without interrupt:

Here is disassembled code which cause problem:

    1046: 0804507b           lp.setupi x0,128,1056 <_loop_end>

0000104a <_loop_start>:
_loop_start():
    104a: 080000ef           jal ra,10ca <function_sw_data>
    104e: 37e000ef           jal ra,13cc <function_lw_data>
    1052: 77c000ef           jal ra,17ce <function_clear_data>

00001056 <_loop_end>:
_loop_end():
    1056: 10020213           addi tp,tp,256 # 1100 <function_sw_data+0x36>

0000105a <_exit>:
_exit():
    105a: a821                 j 1072 <function_pass>

When we tried to run it I got a lot of error messages because ri5cy skipped execution of the last instruction of the loop.

Here is screenshot of the waveform:

 
[image]

Here we can see that after execution instruction with PC 0x18D0 we see execution of instruction with PC 0x104A which is beginning of the next loop iteration.
Instruction with PC 0x18D0 is the jalr statement of the function "function_clear_data". Instruction with PC 0x1056 (last instruction of the loop) was skipped.

Please note that during this simulation, no IRQ was generated.
this looks like a similar bug to issue #107 but even without IRQs.

for DSPG use, link to our simulation:
lbrun -C simvision -p high -- simvision -64bit -waves /home/dtmp/run/dbm10l/denisg/ri5cy/ws_dbm10l_initial/hw_loop_bug_no_irq/vlg_sim/waves.shm

davideschiavone, 2019-09-19:
Can you please try again this test with this fix?
https://github.com/pulp-platform/riscv/commit/78c4c448bdc7cb0094c2d759ca5352236e7e44c6

best
Davide

davideschiavone, 2019-10-11:
Fixed by https://github.com/pulp-platform/riscv/commit/f19f0722b06b53a59cb38ea982d909f2d8797d13

davideschiavone, 2019-11-05:
The solution proposed contains a combinational loop,
please find the proposed fix for that combinatorial loop here:
https://github.com/pulp-platform/riscv/issues/190

I will merge to master once I get feedback from your functional verification and sorry for inconvenience

yaronbe1, 2019-11-11:
not yet. verification team is on it.

yaronbe1, 2019-11-11:
fix looks OK
you can commit

davideschiavone, 2019-11-12:
Thanks @yaronbe1

closed 2019-11-12 by davideschiavone