openhwfoundation/cv32e40p #162: Simple setup of ri5cy core + adv_dbg_unit + jtag

aignacio, 2019-09-19:
Hi everyone,
is there some kinda of simple setup of ri5cy core + adv_dbg_unit + jtag interface available? I'm trying to make a testbench using verilator and I have all the setup to connect the jtag signals over the core to test but I couldn't find any non-interconnect dependent example of how to make it happen straightforward...seems that all wrapper that I found dependent upon some interconnect to send the debug commands, maybe I missing the right flow anyway tks :godmode: 
tks!

bluewww, 2019-09-19:
There is a testbench [here](https://github.com/pulp-platform/riscv/tree/master/tb/dm) that uses RI5CY, the debug module and a jtag interface, accesible over OpenOCD.

aignacio, 2019-09-19:
Hi @bluewww,
tks for the answer, in my case where I'm not going to use DPI interface, is that need to instanciate the [SIMJtag](https://github.com/pulp-platform/riscv/blob/master/tb/dm/tb_test_env.sv#L304) or can I use the connections directly from dmi_jtag module?

bluewww, 2019-09-19:
The SimJTAG is to driver the jtag pins over DPI. If you don't want to drive them like this you can remove it and do it another way (e.g. if you use verilator you could use c++ directly to control the jtag signals).

aignacio, 2019-09-19:
Yes, that's my idea. Tks!

closed 2019-09-19 by aignacio