openhwfoundation/cv32e40p #164: Cannot debug through gdb + verilator

aignacio, 2019-09-19:
Hi @bluewww,
I was using the testbench that you mentioned on the issue #162 and I was able to connect with openocd and rbb protocol through the target where it can found the jtag tap but for some reason it dumps on verilator model that its fetching illegal instruction at addr 0x00 like in the image below
[image]

So after some time the gdb asnwer with the following response:
```
Reading symbols from output/test.elf...
Remote debugging using :3333
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response

Ignoring packet error, continuing...
Remote replied unexpectedly to 'vMustReplyEmpty': timeout
```
I changed a bit the openocd script file but I don't know that this could lead to the problem..
```
#debug_level 4
adapter_khz     10000

interface remote_bitbang
remote_bitbang_host localhost
remote_bitbang_port 8080

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x249511C3

foreach t [jtag names] {
    puts [format "TAP: %s\n" $t]
}

set _TARGETNAME $_CHIPNAME.cpu
#target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0x3e0
target create $_TARGETNAME riscv -chain-position $_TARGETNAME -rtos auto

riscv set_reset_timeout_sec 2000
riscv set_command_timeout_sec 2000

# prefer to use sba for system bus access
riscv set_prefer_sba on

# dump jtag chain
scan_chain

init

halt
echo "Ready for Remote Connections"
```
I'm using the bootrom provided in the same folder as well and my tb in verilator does not do anything besides resetting the core and opening the tcp server for the jtag connection, once the commands arrive from openocd, I latch the pins and tick the simulation. Attached also there's my top wrapper + verilator tb.
[top.zip](https://github.com/pulp-platform/riscv/files/3632331/top.zip)

bluewww, 2019-09-19:
Did you check the instruction trace log? There might be useful information there.

aignacio, 2019-09-19:
What log you refer to? Do I need to enable the define TRACE_EXECUTION macro to see this log?

bluewww, 2019-09-19:
>  Do I need to enable the define TRACE_EXECUTION macro to see this log?

Yes. Afterwards you should see a `log` file in the folder where you executed the simulation

aignacio, 2019-09-20:
I tried to compile with this macro although looking to the RTLs seems that's not compatible with verilator (`ifndef VERILATOR `ifdef TRACE_EXECUTION) and forcing to add in the compilation, verilator complain about the system verilog as well.

aignacio, 2019-09-20:
Looking to the verilator output there's some error on the instruction received in debug memory map:
[image]
[image]

aignacio, 2019-09-20:
I was using the openocd build from **mainline** now I change to riscv-openocd and once I try to connect the debug log from openocd keeps dumping the same log:
Debug: 3212 59663 riscv-013.c:408 scan():  ->  allresumeack anyresumeack allhalted anyhalted authenticated version=2
```
Debug: 3213 59663 riscv.c:1739 riscv_openocd_poll():   no harts just halted, target->state=2
Debug: 3214 59763 riscv.c:1721 riscv_openocd_poll(): polling all harts
Debug: 3215 59763 riscv.c:2632 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 3216 59763 riscv.c:2648 riscv_invalidate_register_cache(): [0]
Debug: 3217 59763 riscv.c:1675 riscv_poll_hart(): polling hart 0, target->state=2
Debug: 3218 59764 riscv-013.c:397 scan(): 41b 5i r 00000000 @11 -> + 00000000 @11
Debug: 3219 59807 riscv-013.c:397 scan(): 41b 5i - 00000000 @11 -> + 00030382 @11
Debug: 3220 59807 riscv-013.c:408 scan():  ->  allresumeack anyresumeack allhalted anyhalted authenticated version=2
```

bluewww, 2019-09-20:
I'll have a look at `tb/dm` maybe we recently broke something

aignacio, 2019-09-26:
Hi @bluewww,
Just to inform you, I decided to extend the timeout value and after waiting something like 3 or more minutes by each gdb command, seems that the core it's receiving and responding correctly, but it's **very slow**...what in theory makes impracticable to use for debugging programs. The interface that I'm using with the core uses a tcp server to handle the commands but I cannot believe that should be so slow to drive it, what are your experience with practical usage with VPI driver over riscv-openocd?

Ps.: I choose rbb cause rocket-chip uses this approach and using small/medium cores doesn't look to be so slow with verilator in the same host machine.

bluewww, 2019-09-26:
Do you have patches to make `tb/dm` work with verilator?

I know with commercial simulators gdb commands are also taking ages which is why I only run that setup (gdb + simulator) in a batch fashion. Running all gdb tests in `riscv-tests` takes about half a day. It is weird though that rocket chip doesn't struggle with that setup so I suppose there could be somewhere a performance issue with our setup.

aignacio, 2019-09-26:
@bluewww for sure, I'll try to dig around what I understand of the debug module, anyway, I can create a patch/pull request with the files to make verilator work with rbb in the tb/dm, the only problem is that my [testbench](https://github.com/aignacio/ri5cy_cluster/blob/master/src/jtag_verilator/cluster_jtag_tb.cpp#L180) is a bit different from yours, so if you would like to accept something like this in the future I'll do it for sure  :smiley:

closed 2019-09-26 by aignacio