openhwfoundation/cv32e40p #167: RI5CV Post Synthesis Simulation - Hart Halt !

ranaya123, 2019-09-21:
Hi All,

This question is regarding the gate level simulation of RI5CY core on Pulpissimo platform. Right now I only want to perform this simulation on core level. So I have the synthesized netlist of the riscV core, added it to the build tree with the model file of the standard-cells, and successfully built the system.

When I run the simulation in Modelsim, I can see that the nestlist of the riscV core appears in the design hierarchy and the SDF file was properly annotated to the right scope :

**tb_pulp.i_dut.soc_domain_i.pulp_soc_i.fc_subsystem_i.FC_CORE.lFC_CORE**

Until JTAG propagates the debug request signal, everything is fine : 
-----------------------------------------------------------------
 [TB]       1ns - Using FLL
 [TB]       1ns - Not using CAM SDVT
 Loading default stimuli
[JTAG] SoftReset Done(    701ns)
[JTAG] Bypass Test Passed (  33301ns)
[JTAG] Tap ID: 249511c3 (  43701ns)
[JTAG] Tap ID Test PASSED (  43701ns)
[test_mode_if]   50301ns - Init
[TB]   50301ns - Enabling clock out via jtag
[test_mode_if]   51801ns - Setting confreg to value 003.
[TB]   51801ns - jtag_conf_reg set to 003
[TB]   51801ns - Releasing hard reset
[TB]   53401ns - Init PULP TAP
[pulp_tap_if] WRITE32 burst @1c008080 for           4 bytes.
[TB]   67501ns - Write32 PULP TAP
[JTAG] R/W test of L2 succeeded
[TB]  177701ns - Halting the Core
-----------------------------------------------------------------

From this point onwards, the core seems to be stucked at Halt mode. After diving bit deeper into the simulation, the cause of "Halt" appears to be a wrongly issued instruction address caused by the FSM in the prefetch_buffer_i. Take a look at the following figure for RTL simulation :
[image]

At the time instance where the "white arrow" is, **branch_i** perfectly becomes zero at the rising edge of the clock. So that, according to riscv_prefetch_buffer FSM, the **instr_addr_o = fetch_addr** when the **CS=WAIT_RVALID**. But in synthesized design, with realistic delays, branch_i never becomes 0 at this time instance as shown in following figure:
[image]

The combinatorial delay of FSM slightly stretches **branch_i** and as a consequence, **instr_addr_o = addr_i** instead of  **fetch_addr** as shown. This affects the CS and NS as well. From this point onwards, **instr_addr_o** stucks at **1a110804** which is not the intended behavior. 

Since **branch_i** is produced at **riscv_if_stage** wrt to the rising edge of the clk and checking its status again at **prefetch_buffer** in the same clk cycle won't result the same RTL level simulation outputs.

Any input would be greatly appreciated...
Thanks in advance

gautschimi, 2019-09-23:
the problem is most likely at the interface of the core and the instruction cache/memory. one side is sdf annotated and I guess there are no annotations on the cache side. This will cause the problems. I suggest to look at all the instr_** signals of the core interface and double check if the delays do not violate the provided constraints.

later on in the simulation you will face the same problem on the data interface.

gautschimi, 2019-09-23:
in your case I think the instr_rvalid_i arrives in the wrong clock cycle. on the cache side it changes its value with the rising edge of the clock. and it is sampled in the core with the rising edge of the clock. the problem is that the clock of the core has sdf annotations and will be delayed with respect to the clock of the cache -> this will result in a heavy hold time violation.

you have to annotate the full design with timings, or properly constrain the interfaces. if you don't annotate the sdfs you will see that the simulation is going to work

ranaya123, 2019-09-30:
Thanks for your inputs @gautschimi . I solved the issue. The trick is to replace the clock gating cells (applied by the synthesizer) by their behavioral models, so that data is sampled at exactly the same clock edge from both peripheral and core ends. So now the simulation is working just fine.....

Regards
Anuradha

closed 2019-09-30 by ranaya123