openhwfoundation/cv32e40p #169: Illegal Instruction Exception not Raised - Dynamic Rounding Mode

shetalani, 2019-09-27:
**RISC-V Specification:**

- _"Floating-point operations use either a static rounding mode encoded in the instruction, or a dynamic rounding mode held in frm."_ 
- _"Rounding modes are encoded as shown in Table 11.1. A value of 111 in the instruction’s rm field selects the dynamic rounding mode held in frm."_
- _"If frm is set to an invalid value (101–111), any subsequent attempt to execute a floating-point operation with a dynamic rounding mode will raise an illegal instruction exception."_

**Issue Description:**

An illegal instruction exception is not raised for the case when a floating-point instruction uses a dynamic rounding mode while the frm is set to an invalid value. 

**Example:**

As shown below, the instruction 32'hc011f1d3 (fcvt.wu.s x3, f3) is decoded at time point t##0 where rm field has the value 111, and frm has the value 101 (written by a previous instruction), with no illegal instruction being flagged, as illegal_insn_dec is de-asserted.

[image]

---

Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2019.2.2

stmach, 2019-10-08:
The issue seems to be that the pipeline is not flushed upon CSR write to frm, i.e. the instruction decoded at t##0 does not see the updated CSR (and should actually stall until t##1).
We'll have a look.

pascalgouedo, 2023-10-09:
Resolved with PR #801

closed 2023-10-09 by pascalgouedo