openhwfoundation/cv32e40p #170: Illegal Instruction Exception not Raised - FS Field

shetalani, 2019-09-27:
**RISC-V Specification:**

- _"The FS field encodes the status of the floating-point unit, including the CSR fcsr and floating-point data registers f0–f31"_
- _"In systems that do not implement S-mode and do not have a floating-point unit, the FS field is hardwired to zero."_ 
- _"When an extension’s status is set to Off, any instruction that attempts to read or write the corresponding state will cause an illegal instruction exception."_

**Issue Description:**

Accessing the F-extension CSRs / floating-point data registers f0–f31 while the FS field of MSTATUS is set to OFF doesn't raise an illegal instruction exception.

**Example:**

As shown below, the instruction 32'h23aaf3 (csrrs x21, frm, x7) is decoded at time point t##0, while FS field is set to OFF, with no illegal instruction being flagged, as illegal_insn_dec is de-asserted.

[image]

---

Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2019.2.2

EkanshBhatnagar, 2022-05-04:
Can I tackle this bug ?

davideschiavone, 2022-05-04:
hi @EkanshBhatnagar , of course you can, everyone can :) thanks, please synchronize with @pascalgouedo

pascalgouedo, 2022-05-04:
Hi,
After some CV32E40P RTL analysis, here is the status:
- FS (and SD) are always hardwired to 0 even when FPU = 1

After some RISC-V spec analysis, here are some remarks:
- If no S-Mode and FPU = 0 (v1 tag case), F-extension CSRs / f0–f31 registers read/write accesses and FS field write access should raise illegal exception => agreed.
- If no S-Mode and FPU = 1 (will be v2 tag case), FS should be writeable by csr instruction and modifiable to Dirty when F-extension CSRs / f0–f31 registers are written/updated => Not done in present v1 RTL. New issue to create.

pascalgouedo, 2023-10-09:
Resolved with PR #801

closed 2023-10-09 by pascalgouedo