openhwfoundation/cv32e40p #174: Incorrect F Extension Result Calculation - Dynamic Rounding Mode

shetalani, 2019-09-30:
**RISC-V Specification:**

- _"Floating-point operations use either a static rounding mode encoded in the instruction, or a dynamic rounding mode held in frm."_
- _"Rounding modes are encoded as shown in Table 11.1. A value of 111 in the instruction’s rm field selects the dynamic rounding mode held in frm."_
- _"All floating-point to integer and integer to floating-point conversion instructions round according to the rm field."_

**Issue Description:**
A wrong value is written to the normal RF for an F-extension instruction that has a dynamic rounding mode.

**Example:**

As shown below, the instruction 32'hc00f77d3 (fcvt.w.s x15, f30) is decoded at time point t##0 where rm field has the value 111, and frm has the value 011 (written by a previous instruction). Where f30 has the value 32'h02000000 already by a previous instruction as well. Hence, the result has to be rounded up according to the value of frm. However, the rounding mode signal "...fpnew_bulk.rnd_mode_i" has the value associated with RNE and x15 is written with the value 0 instead of the right result 1. 

**Apparently, ri5cy doesn't support dynamic rounding mode! Is that right?** 

[image]

---

Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2019.2.2

stmach, 2019-10-08:
This bug is related to #169 where the change to `frm` in the previous cycle is not seen by the `fcvt` instruction following it.
Leaving this open until #169 is closed.

pascalgouedo, 2023-10-09:
Resolved with PR #801

closed 2023-10-09 by pascalgouedo