openhwfoundation/cv32e40p #175: Incorrect F Extension Result Calculation -

shetalani, 2019-09-30:
**Issue Description:**

A wrong value is written to the floating point RF for an F-extension instruction.

**Example:**

As shown below, the instruction 32'hf9ef2d43 (fmadd f26,f30,f30,f31) is decoded at time point t##0, where f30 has the value 32'h2b909000 and f31 has the value 32'h80000000 (written by previous instructions). Incorrectly, the value 32'h17420944 is written to the register f26 instead of the valid result 32'h17a344a2. 

[image]

---

Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2019.2.2

stmach, 2019-10-08:
Hello, testing the FPU itself with the given input values produces the valid result, so I would need you to provide a bit of further information:
What are the operands (`operands_i`) that are provided to the unit at `t##1`? Unfortunately in your screenshot the signal is collapsed.

shetalani, 2019-10-08:
If I recall correctly, the operand values sent to the unit were expected. Here are screenshots for FADD and FSUB with expanded operand_i:
 
[image]

[image]

stmach, 2019-10-08:
Very strange, I am not able to reproduce the issue on RTL simulation or using Verilator. All inputs to the FPU are correct, the result coming out just seems wrong which doesn't happen with vsim or Verilator.
We had reports of the FPU returning wrong results on post-synthesis netlists, that however was due to actual bugs in an older version of a vendor's synthesis tools.
Could you recheck using e.g. Verilator to see if it returns the same values as in your simulation run?

shetalani, 2019-10-09:
Here is a testbench that you could use to reproduce the following counter example: 

[image]

[testbench_issue175.zip](https://github.com/pulp-platform/riscv/files/3707086/testbench_issue175.zip)

Hope this helps.

stmach, 2019-10-09:
Thank you for the testbench. Running it on RTL sim (vsim) yields the correct results as highlighted here:
[image]

I am using latest master branch (4344de8fd37874ad0b392e90630b647763f96037), the only change made from default is enabling the FPU of course.

shetalani, 2019-10-11:
It is really weird, I set the parameters as you did and still getting the same issue. 

I am using this filelist to read in the design, is there something I am missing here?
[flist.zip](https://github.com/pulp-platform/riscv/files/3717488/flist.zip)

Could you share your simulation setup with me.

stmach, 2019-10-14:
In your file list, the dependencies of `fpnew` from `common_cells` are unrolled into a directory called `rtl`. This is per se not bad, just so you know the only two source files you need from the `common_cells` dependency are:
```
fpnew/src/common_cells/src/lzc.sv
fpnew/src/common_cells/src/rr_arb_tree.sv
```
I am assuming you're using either the latest release tag of the fpu or the `develop` branch there (I tested both and they work).

For my quick test, I used the following (mostly manual) setup:
- I enabled the simulation of the FPU in the core testbench, you can check out the `enable-fp-tb` branch for that. We'll be merging this into master as soon as things are a bit more cleaned up but for this simple test it should be enough.
- In the testbench file you provided (`testbench_issue175.sv`), properly enable the FPU through the parameter on the core instance by changing line 4326 to:
```SystemVerilog
riscv_core #(.FPU(1)) DUT(
```
In your setup it seems you changed this parameter on the `riscv_core` module itself which works as well, but I'm going from what's in the repo.
- Go into the `tb/core` directory
- Run `make vsim-all` which will analyze and optimize the core RTL (including FPU, check Makefile or console output for the list of files analyzed).
- Open vsim (or run `make vsim-run-gui` to open the gui)
- In vsim, analyze and simulate your provided testbench file (with full visibility)
```tcl
vlog -sv testbench_issue175.sv
vsim TB_riscv_core_RV_chk_ops_RV32F_RI5CY_all_a -voptargs=+acc
```
- Add some waves `add wave -position insertpoint sim:/TB_riscv_core_RV_chk_ops_RV32F_RI5CY_all_a/DUT/ex_stage_i/genblk1/genblk1/i_fpnew_bulk/*`
- Run it and observe the waves: `run -all`. The console output is unfortunately quite a mess in the vsim window.

shetalani, 2021-03-10:
Hi @stmach, thank you for the detailed response, I am closing this ticket as the bug reported here is rooted to my local wrong setup.

closed 2021-03-10 by shetalani