openhwfoundation/cv32e40p #176: Problem on second requisition on misaligned stores

aignacio, 2019-10-02:
Hi everyone,
I have been designing a dbus to ahb converter for RI5CY cpu and I have some questions related to set the converter to operate with two-cycle misaligned requisitions. So, considering that a memory or cache interface could handle misaligned requisitions through ahb bus, how did you make the conversion in terms of AHB Address considering the two LSBs  from the req/grant interface on the RI5CY core? I mean, as an example here below, have a word to store shifted by one byte into (0xb000_0000), so the first req. makes sense ahb addr == addr_i but in the second requisition to store the remaining byte, the addr_i it's equal to 'h5, what indeed would shift the data by one byte once you make ahb_addr == addr_i, IMO the addr should be 'h4 (nxt word boundary), isn't it? 
```
    li t0, 0xDEADBEEF
    li t1, 0xB0000000  
    nop
    sw t0, 0(t1)        # 0xB0000000 = DEADBEEF /  0xB0000004 = 00000000
    nop
    sw t0, 1(t1)        # 0xB0000000 = ADBEEFEF  /  0xB0000004 = 000000DE
```
[image]

Silabs-ArjanB, 2020-05-07:
Hi @aignacio, I agree with your observation and I had actually filed as similar ticket (#318) to ease conversion to protocols as AHB and AXI. If this pull request is accepted (https://github.com/openhwgroup/cv32e40p/pull/322) then the issue will have been fixed as you suggested.

Silabs-ArjanB, 2020-06-15:
Hi @aignacio This issue has been fixed now (with above pull request). If you agree can you please close this issue?

aignacio, 2020-06-15:
yeap, done

closed 2020-06-15 by aignacio