openhwfoundation/cv32e40p #177: hw loop initialization

yaronbe1, 2019-10-02:
we have found  potentially a bug. Test flow:

•	hw loop zero start and end registers were configured.
•	before hw loop zero initialization (lp cnt 0 at addr 0x8000_0186) an irq was set
•	lc cnt 0 instruction was not executed due to irq flush but the lpcount0 csr register was set regardless the irq flush
•	the irq handler wrote zero to lp count0 csr reg (0x7c2 csr addr)
•	after mret lpcount 0 instruction was executed and lpcount0 csr reg set to 5

see in waves:
[image]

We can see the potential bug here in such a scenario. If the irq handler needs to use a hw loop at first it might check if there any free hw loops. The hw loop is available if the hw cnt is zero. as we can see in the situation above the hw cnt can be more than zero even if the hw loop is not occupied yet.

We think maybe HW should apply restrictions to some operations on irq flush e.g. setting value to hw cnt csr register.

for DSPG internal: wave at: /home/dtmp/run/dbm10l/oleksa/ri5cy/ws_dbm10l_initial/hw_irq_dbg

dspg-arch1, 2019-10-16:
More information, may help define and fix:

the issue we found shows that hw loop counter basically set two times per instruction - before id stage, and during id stage. This is potentially algorithmic bug regardless the interrupt. In the example we used interrupt for two purposes:
 
1 - it is easy to split and show hw loop cnt behavior 
2 - this is first idea of algorithm breaking we can imagine
 
I am not sure if we can avoid all algorithm breaking situations by avoiding hw loop in certain situation. **We would prefer to have a fix to have a register set only during id stage.**
 
Keeping this bug unsolved gives next issues for SW from my perspective:
•	relevant information about HW loop status is missed in IRQ handling procedure. so it is impossible to check that some HW loop is free and it can be used in the IRQ handler
•	impossible to terminate existing HW loop by setting counter to 0 (break instruction) or go to next iteration of the loop using it start address (continue instruction) from the IRQ handler
 
We implemented in the scoreboard comparison of all GPR and CSR registers between design and the spike after execution of each instruction.
Because of problems with termination of HW loop initialization instruction by IRQ we have tomes of error messages like "HW loop start address mismatch" and etc.
This error is caused by difference in behavior of spike and design: spike terminated HW loop initialization instruction, but design is not.

davideschiavone, 2019-10-22:
By the way, the HWLoop registers need to be stored into the STACK in interrupt service routines.
They are part of the context, so if the ISR needs to use HWLoop it does not have to check anything, it simply stores in the stack all the HWloop CSRs and then use them normally.
Before mret, it must restore them.

dspg-arch1, 2019-10-24:
There is an lp_setup instruction at address 0x8000_00186 and there is an interrupt taken "at the same time"

The MEPC registers is set to 0x8000_0186 which implies that the instruction was cancelled, the spike model predicts this cancellation.

See from the waves that the lp_cnt[0] is set which implies that the instruction was executed.

Since the MEPC will re-execute the instruction as 0x8000_0186 we think the instruction at the entrance to the interrupt routine should have been cancelled to avoid the double execution, not to cancel the instruction at the return from the interrupt handler.

[image]

davideschiavone, 2019-11-12:
I guess this has also been fixed by ad00d1f4077b1192a80dc8232bc76ee72fb37c91 ? @yaronbe1

yaronbe1, 2019-11-13:
> External Email - Use caution when responding, clicking and/or downloading attachments.
>
> I guess this has also been fixed by ad00d1f<https://github.com/pulp-platform/riscv/commit/ad00d1f4077b1192a80dc8232bc76ee72fb37c91> ? @yaronbe1<https://github.com/yaronbe1>
>
>
> DSP Group, Inc. automatically scans all emails and attachments using MessageLabs Email Security System.
>
> DSP Group, Inc. automatically scans all emails and attachments using MessageLabs Email Security System.

Yes looks like the fix to 148 fixes also this one

closed 2019-11-13 by davideschiavone