openhwfoundation/cv32e40p #185: Exceptions update CSRs while in Debug Mode

shetalani, 2019-10-11:
**RISC-V Specification:**

- _"When executing code from the optional Program Buffer, the hart stays in Debug Mode and the following apply: Exceptions don't update any registers. That includes cause, epc, tval, dpc, and mstatus. They do end execution of the Program Buffer."_

**Issue Description:**

While in debug mode the instruction ECALL updates some CSRs, while it should not. 

**Example:**

As shown below, the instruction 32'h73 (ecall), decoded at time point t##0 where the current privilege is M and debug_mode is set, updates the MEPC, MCAUSE and MSTATUS registers.

[image]

---
Product: OneSpin 360 DV-Verify
App: RVV
Tool's version: 2019.2.2

bluewww, 2019-10-11:
* _"Almost all instructions that change the privilege level have **undefined behavior**. This includes
ecall, mret, sret, and uret. (To change the privilege level, the debugger can write prv in
dcsr). The only exception is ebreak. When that is executed in Debug Mode, it halts the
hart again but without updating dpc or dcsr"_
According to the spec and my interpretation therof the behaviour of `ecall` is undefined so anything can happen. Don't use `ecall` in the program buffer.

jm4rtin, 2019-11-11:
When looking at why new versions of the J-Link GDB server stopped working we noticed a related issue. The GDB server would tell GDB that there were FPU registers even though the core was instantiated without the FPU. This is a separate issue but it exposed a problem with RI5CY. This would cause GDB to attempt to read the FPU registers. It did this using the program buffer and a fmv.x.w instruction. This caused an illegal instruction exception.

When causing an illegal instruction exception in the program buffer it follows the normal exception processing like in machine mode. It can leave the core in a crippled state.

Silabs-ArjanB, 2020-07-15:
Hi @shetalani This issue has been fixed with https://github.com/openhwgroup/cv32e40p/pull/353. If you agree can you please close this issue; if not, please let us know what issue you are still seein.

Silabs-ArjanB, 2020-07-23:
Hi @shetalani We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-23 by Silabs-ArjanB