openhwfoundation/cv32e40p #186: cornercase situation lp.setupi + IRQ

dspg-arch1, 2019-10-15:
This may be similar to Issue 177

Here we can see IRQ with ID = 0x7 was activated and as the result instruction with PC = 0x80000336 was canceled.
I did this conclusion because signal "id_valid" become 0 on the ID stage of instruction execution.
But register hardware loop counter #0 changed its value from 1 to 4. 
So not valid ID stage of "lp.setupi" instruction affected system by changing "lpcount[0]" register. 
Such register update was not expected for spike simulator.
[image]

Actual Error Message:
ri5cy/verif/classes/pulpino_scoreboard_c.sv(384) @ 328493000: uvm_test_top.pp_env.pp_scoreboard [SPIKE_DBG] HW loop count #0 compare mismatch. in the spike 0x1 -> 0x1, in the design: 0x4

davideschiavone, 2019-10-16:
Hello everybody,

to help me reproduce the issue, can you provide a kind of testbench looking like this one please?

https://github.com/pulp-platform/riscv/files/3707086/testbench_issue175.zip

Thanks a lot
Davide

yaronbe1, 2019-10-17:
Unfortunately we do not have a verilog stimulus.
We do have the ASM file of the test.please find bellow:
[riscv_asm_test.0.S.zip](https://github.com/pulp-platform/riscv/files/3739283/riscv_asm_test.0.S.zip)

yaronbe1, 2019-10-17:
We have also prepared a simpler firmware in which interrupts are enabled only when lp.setupi instruction is executed.
[riscv_fm_asm_test.0.S.zip](https://github.com/pulp-platform/riscv/files/3739329/riscv_fm_asm_test.0.S.zip)

When you run it with enabled IRQ's with any seed you will get error like: 
UVM_ERROR /proj/dbm10l/denisg/ws_dbm10l_test/blk/ri5cy/verif/classes/pulpino_scoreboard_c.sv(425) @ 7663000: uvm_test_top.pp_env.pp_scoreboard [SPIKE_DBG] HW loop count #0 compare mismatch. in the spike 0x0 -> 0x0, in the design id: 0x1-> ex:0x1

Test pass criteria is execution of "function_pass" routine.

davideschiavone, 2019-10-22:
Hello Yaron,

In this piece of code you sent me (the last one) I see you are comparing the counter of the hwloop against 0, but that is wrong because the counter is not updated internally during the last iteration. 
In my waveforms I don't see the counters changing.

I have my hwloop setup being interrupted, it writes the counters then the core jumps to the ISR and storing in the MEPC the hwloop PC (which is wrong as the instruction will be executed again once the MRET returns from interrupt).  In this specific ase (lp.setupi), the re-execution won't lead to errors (although it is conceptually wrong and I am fixing it) as it will simply set-up the HWLoop again. But the counters aren't changed.
Here you see the trace of my execution.
The core is enabling the interrupts, jumping to them (ISR contains only JAL and MRET) and then it setups the HWLoop. Here you don't see the in the trace that actually before the JAL the core already sets the HWLoop, but the trace is not catching it due to interrupts.

> 24956ns            2491 0000013e 30045073 csrrwi           x0, 0x00000008, 0x300
>   25036ns            2499 0000007c 3300006f jal              x0, 816            
>   25056ns            2501 000003ac 30200073 mret                                
>   25106ns            2506 00000142 0015d07b lp.setupi        0x0, 0x1, 0xb      
>   25116ns            2507 00000146 30005073 csrrwi           x0, 0x00000000, 0x300
>   25156ns            2511 0000014a 7c2022f3 csrrs            x5, x0, 0x7c2         x5=00000001

I read the counters as 1.

Maybe I did not understand the problem?

Itaykeidar, 2019-10-22:
Hi,
the code attached was just to emphasize the issue, perhaps the comparison to "0" was not correct.
the original code if i am not mistaken (was looking into our bug tracking) is attached.

also, your trace does not show where an interrupt is occurring, which is also important.

[irq_on_lp_setupi.zip](https://github.com/pulp-platform/riscv/files/3756080/irq_on_lp_setupi.zip)

Regards,
Itay K.

Itaykeidar, 2019-10-23:
Hi David,
Hopefully i amnot mistaken, but i reasked the team and this test is not intended to have a HW loop inside the irq routine.
The problem is that the HW regs change while.entering the irq routine, without going into another HW loop.
A question regarding the hw loop regs push to stack - is the compiler doing it automatically when converting a C code loop to asm hw loop?

davideschiavone, 2019-11-05:
Just to have it written for tracking reasons, 

I provided two solutions and waiting for feedback before merging to master.
Those 2 solutions are:
1) ad00d1f4077b1192a80dc8232bc76ee72fb37c91
This prevents the HWloop instruction being executed if the interrupt is taken to be aligned with the other instructions

2) af500316da0bc9ce8cb51bdf44ed407e76e0485d
This is delaying the interrupt taken to first execute the instruction in the ID stage. We are more in favor for 1) but waiting for functional verification feedbacks

yaronbe1, 2019-11-11:
ad00d1f fix seams good.
you can commit.

davideschiavone, 2019-11-12:
Thanks @yaronbe1

closed 2019-11-12 by davideschiavone