openhwfoundation/cv32e40p #187: Different behavior of spike and design on c.jr zero instruction

dspg-arch1, 2019-10-22:
On processing this instruction, design performs jump operation to address 0x0. We can see it on the waveform.
But implementation of this instruction in the spike has additional require statement  which marks it as illegal instruction.
The link to c.jr instruction implementation in the github is: https://github.com/riscv/riscv-isa-sim/blob/master/riscv/insns/c_jr.h (shown here)
   1 require_extension('C'); 
   2 require(insn.rvc_rs1() != 0); <<<< true?
   3 set_pc(RVC_RS1 & ~reg_t(1)); 

 
What is the correct behavior  in this case? 
•	perform jump instruction to PC 0x0
•	mark this instruction as illegal.

bluewww, 2019-10-22:
`c.jr x0, 0` is reserved. We need to raise an illegal instruction exception.

Silabs-ArjanB, 2020-05-19:
We should treat this as an illegal instruction as we should not add 'extensions' in space that is reserved by the RISC-V organisation.

Silabs-ArjanB, 2020-06-02:
Hi @dspg-arch1 This issue has now been fixed; could you please close this if you agree. The issue will automatically be closed June 10 2020.

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.

closed 2020-07-14 by Silabs-ArjanB