openhwfoundation/cv32e40p #191: Hw Loop initialization (another case)

dspg-arch1, 2019-11-06:
It seems that HW loop registers were applied before ID stage of the instruction in a new case.

[image]

Maybe some speculations:
In this test we can see  reported before issue with HW loop setup instruction with IRQ.
Probably the reason why it happen is caused by instruction executed before lp.setupi. As I know execution stage rem instruction is quite long.
So ID stage of lp.setupi instruction triggers when EX stage of previous instruction is completed.
But in the waveform we can see that new values of HW loop registers were applied too quickly.

davideschiavone, 2019-11-06:
Hello ,

which fix are you using? (which branch)
Thanks
Davide

yaronbe1, 2019-11-06:
this one:
1.	ad00d1f
This prevents the HWloop instruction being executed if the interrupt is taken to be aligned with the other instructions

davideschiavone, 2019-11-06:
Please apply to that also the next commit:
https://github.com/pulp-platform/riscv/commit/eb07d277b462f227dae482dec987a34181d3f1a6

Let me know if this solves the issue

yaronbe1, 2019-11-11:
fix seams good.
you can commit.

davideschiavone, 2019-11-12:
Thanks @yaronbe1

closed 2019-11-12 by davideschiavone