openhwfoundation/cv32e40p #192: last instruction of HW loop body is skipped

dspg-arch1, 2019-11-06:
There is  HW loop #0 configured to run 10 times following 2 instructions: "jal ra, _some_function" and "add x20, x20, x21".
Instruction jal performs jump operation to address with mark "_some_function" and stores NPC value to register ra. In our case NPC should have pointer to the instruction "add x20, x20, x21". Pointer "_some_function" stores address of some block of code with dummy calculations that ends with instruction "jalr ra". This instruction performs jump operation with the address that stored in ra register.
So we can predict that HW loop should execute 10 times next sequence of instructions: jump to "_some_function", perform dummy instructions, jump back to the loop, do add instruction.

After  jumping back to the HW loop, the last instruction "add x20, x20, x21" was skipped.

Source code of the test:

_start_jmp:
    j _start   #8000 0080
_start:

    li x20, 1
    li x21, 1

    lp.starti x0, __hw_loop_start
    lp.endi  x0, __hw_loop_end
    lp.counti x0, 10

__hw_loop_start: jal ra, _some_function
#     add x21, x21, x20
__hw_loop_end:  add x20, x20, x21

    j _start_jmp

_some_function:
    nop
    nop
    nop
    nop
    nop
    nop
    jalr ra

To simplify debugging I will add few lines from dissemble file..

"_some function" block of code:

8000009a <_some_function>:
_some_function():
8000009a: 0001                 nop
8000009c: 0001                 nop
8000009e: 0001                 nop
800000a0: 0001                 nop
800000a2: 0001                 nop
800000a4: 0001                 nop
800000a6: 9082                 jalr ra

HW loop initialization with body:
80000086: 0060007b           lp.starti x0,80000092 <__hw_loop_start>
8000008a: 0060107b           lp.endi x0,80000096 <__hw_loop_end>
8000008e: 00a0307b           lp.counti x0,10

80000092 <__hw_loop_start>:
__hw_loop_start():
80000092: 008000ef           jal ra,8000009a <_some_function>

80000096 <__hw_loop_end>:
__hw_loop_end():
80000096: 9a56                 add s4,s4,s5

In the waveform we can see that after jumping back to the HW loop body, counter was decremented and next executed instruction has PC 0x80000092, but expected PC should be 0x80000096. HW loop counter should be decremented after execution of the add instruction.

Error message from the scoreboard:
UVM_ERROR /proj/dbm10l/denisg/ws_dbm10l_test/blk/ri5cy/verif/classes/pulpino_scoreboard_c.sv(323) @ 2963000: uvm_test_top.pp_env.pp_scoreboard [SPIKE_DBG] PC compare mismatch. in the spike 0xffffffff80000096, in the design: 0x80000092

To confirm that instruction was executed with mistakes, I added register x20 and x21 to the waveform.W we saw they are still constant 1 till the end of the simulation, so add instruction which can modify it was not executed.

davideschiavone, 2019-11-06:
Hello,

can you please tell me which version of the core you are using?

I need to reproduce it, so I need to know which fixes you have included in the RTL

Thanks
Davide

yaronbe1, 2019-11-06:
date: 15OCT2019 commit:  https://github.com/pulp-platform/riscv/tree/aa13a06b4ca5bacdc59a92fd7aff75adb140cd7f

upon that we have implemented the fix to #186   	ad00d1f

davideschiavone, 2019-11-06:
So, I was able to reproduce the bug.

It is quite nasty, basically the jump at the end of the function is jumping back to the HW loop body at cycle 0. There are no stalls in the instruction and data busses. Neither on grants nor on valids.
At cycle 1, the ID stage has no valid instruction, the EX stage is doing X1 = PC+4
The prefetcher now sees that the pc_if contatins the end instruction of the HWLoop, on which the jump instruction jumped to at cycle0.
As there are no stalls, the instruction is valid and the HWLoop controller goes to the state UNALIGNED_COMPRESS. Unfortunatly, at cycle 1 the ID prevents the instruction just fetched in the IF stage (i.e. the last of the HWloop) as it reports a jr_stall, which drivers to 0 the id_ready_o signal.
Although this stall is fake, another situation may happen with long (multy cycles) instructions in the EX stage with for examples HWLoops contains 3 instructions, the last misaligned and compressed.

This is a very nasty bug and I am not sure we can solve it in a reasonable amount of time, altough we can try with some helps.

This situation happens when the last instruction of the HWLoop is misaligned and compressed, is it something that can be prevented at compiler level? @eflamand, @haugoug

eflamand, 2019-11-07:
HW loops are subject to several constraints, some of them coming from the HW some of them SW considerations:

Hardware:
1) Loop must be a perfect loop, e.g loop head strictly dominates all nodes in the loop (you cannot branch from outside the loop to inside the loop without passing through the head)

2) Minimum number of instructions in the loop should be >= 2

3) Last instruction of the loop must not be control related (jump, cond jump, call, csync, rts, rtn)

4) Last instruction of the loop cannot be the target of a branch

All these conditions are checked by the compiler before emitting a HW loop. While this is definitely true for C, C++ or whatever high level language supported by the compiler this is unfortunately not true if you write assembler. In this case no restriction checks are performed and you are on your own ...

Sw consideration: Enabling function call in a loop means that the callee may have to push/pop hw loop CSR registers in case it is using itself hw loop. This is an expensive sequence (cycle/foot print) and in this situation it is better not to use a HW loop. This is also checked by the compiler, when a loop is candidate for hw loop mapping the compiler checks if there are function calls in the loop body, if yes it reject the loop as a candidate for hw loop mapping.

As a general comment hw loops really pays if there are few instructions in the loop (say in between 2 and max 10). Above this threshold the benefit is really questionable.

Eric

dspg-arch1, 2019-11-07:
It is very good the conditions of the loop list above are covered by the compiler, that means we can use compiler generated loops without fixing the bug.

However, the case we are worried about is not:
"Enabling function call in a loop"
but rather
"enabling loop in function call or interrupt routine" where only the compiler can know if there will be HW loop, which one/both of the registers sets will be used, and then to save/restore the state of the registers like any other resource used in function calls and interrupts.
(not part of this github I guess, but can we get an answer if there is a plan to cover this case for the future?)

eflamand, 2019-11-07:
In general an interrupt routine should save all the context including hw loop related registers.
We support an optimized variant where the interrupt handler saves only the registers it is effectively using. It can be activated using the interrupt attribute on a function declaration:
void __attribute__((interrupt)) __rt_timer_handler()

In this case the compiler will check if the handler contains no true function calls (you can have function calls but they have to be locals to the compilation unit since the compiler needs to know exactly what the function does). It also checks that no hardware loops are used in the it handler

dspg-arch1, 2019-11-07:
Our SW architecture will need to use HW loops in interrupt routines, and to save entry and exit time , interrupt latency, we expect to use the optimized variant. We strongly think a HW loop in the handler comes under the classification "registers it is effectively using" as opposed to disabling the HW loop generation or disabling optimization (if i understand correctly). <a phone call may clear this up faster than a back and forth here, suggestions?>

dspg-arch1, 2019-11-11:
THe case we gave had a Call as the instruction before the lp.end, is this case covered by the "jump to end" restriction in the compiler? It is critical for us to know

dspg-arch1, 2019-11-11:
Sorry, it says clearly above that a function call in a loop candidate rejects the possibility of compiler using HW loop. Missed it. Ignore the last comment