openhwfoundation/cv32e40p #197: Prefetch buffer issue when IRQ arrived on last instruction of the HW loop body

yaronbe1, 2019-11-13:
Part of the disassembled code:

800004d4: **0001**                 nop

800004d6 <main_instr_8_st_0_t>:
main_instr_8_st_0_t():
800004d6: b24f05db           p.mulsn a1,t5,tp,25

800004da <main_instr_8_end_0_t>:
main_instr_8_end_0_t():
800004da: 2358**f557**           pv.min.sci.b a0,a7,-21

In this test HW loop #0 was enabled. Start address is 0x800004d6, end address is 0x800004da.

Between iterations of this loop, IRQ arrived.
After returning from IRQ handler by mret instruction prefetch buffer violation occurred.

Screenshot of the waveform:

[image]

In the wavefom there are 2 cursors where 0x800004da was executed and  instructions are different.

Error message in the log file:
UVM_ERROR /proj/dbm10l/oleksa/ws_dbm10l_initial/blk/ri5cy/verif/classes/pulpino_imem_checker_c.sv(40) @ 515973000: uvm_test_top.pp_env.imem_checker [IMEM_CHKR] possible prefetch mismatch, trans instr 0x1f557, resp instr 0x2358f557, pc 0x800004da

for DSPG: To open simulation results use next command line:
/home/dtmp/run/dbm10l/oleksa/ri5cy/ws_dbm10l_initial/prefetch_hwloop/vlg_sim/waves_cmd_file

yaronbe1, 2019-11-14:
we also found similar failing case with loop that has more than 2 instructions in the loop body.
So this issue not depends from the HW loop size.

yaronbe1, 2019-11-14:
this is where the IRQ was taken. you can see it was on the  0x2358f557  instruction:
[image]

davideschiavone, 2020-09-23:
Hi @yaronbe1 , the HWLoop has been completly rewritten from scratch due to this bug :)
Is it still present , or can wel close the issue?

davideschiavone, 2023-08-04:
@pascalgouedo - I think we can close this issue as the HW loop addresses here are misaligned, so it won't apply to todays' CPU, what do you think? or at least waive it

yaronbe1, 2023-08-04:
Hi Davide

I no longer work at DSG Group and to my understanding this activity  was
terminated and the team was laid off, so you can close the case.

Regards
Yaron

davideschiavone, 2023-08-04:
thanks @yaronbe1 for letting us know - I let @pascalgouedo decide whether to close or not this issue

pascalgouedo, 2023-08-04:
Closed due to new hwloops constraints.

closed 2023-08-04 by pascalgouedo