openhwfoundation/cv32e40p #199: Debug Unit issue:  how about the working flow with SSTH

zsr2018, 2019-11-27:
In the previous version, the tb_jtag_debug, what the working method about the unit of Debug_Hit?
Following codes, I can't undersant, why DBG_CTRL_REG need set so many times? 

 task debug_test_single_step;

    logic [31:0] npc;
    logic [31:0] ppc;
    logic [31:0] npc_last;
    logic [31:0] ppc_last;
    logic [31:0] data;
    int i;
    begin
      $display("[DEBUG] Running test_single_step");

      debug_wait_for_stall();
      debug_read(`DBG_NPC_REG, npc_last);
      debug_read(`DBG_PPC_REG, ppc_last);

      debug_read(`DBG_HIT_REG, data);
      if (data[0]) begin
        $display("ERROR: SSTH is set");
        dbg_tb_errors++;
      end

      debug_write(`DBG_CTRL_REG, 32'h0001_0001); // set single-step
      debug_write(`DBG_CTRL_REG, 32'h0000_0001); // RESUME, but Single Step

      for(i = 0; i < 4; i++) begin
        debug_wait_for_stall();
        // read NPC and PPC
        debug_read(`DBG_NPC_REG, npc);
        debug_read(`DBG_PPC_REG, ppc);

        if (ppc !== npc_last) begin
          $display("ERROR: PPC is not equal to last NPC: act %X, expected %X", ppc, npc_last);
          dbg_tb_errors++;
        end

        debug_read(`DBG_HIT_REG, data);
        if (~data[0]) begin
          $display("ERROR: SSTH is not set");
          dbg_tb_errors++;
        end

        npc_last = npc;
        ppc_last = ppc;

        debug_write(`DBG_CTRL_REG, 32'h0000_0001); // RESUME, but Single Step
      end

      debug_write(`DBG_CTRL_REG, 32'h0000_0000); // RESUME, and no Single Step

      //--------------------------------------------------------------------------
      // now the tight branch loop with 2 instructions
      //--------------------------------------------------------------------------
      debug_wait_for_stall();
      debug_write(`DBG_CTRL_REG, 32'h0001_0001); // set single-step

      // do a couple to make sure we are inside the loop
      debug_wait_for_stall();
      debug_write(`DBG_CTRL_REG, 32'h0000_0001); // RESUME, but Single Step
      debug_wait_for_stall();
      debug_write(`DBG_CTRL_REG, 32'h0000_0001); // RESUME, but Single Step
      debug_wait_for_stall();
      debug_write(`DBG_CTRL_REG, 32'h0000_0001); // RESUME, but Single Step
      debug_wait_for_stall();
      debug_read(`DBG_NPC_REG, npc_last);
      debug_read(`DBG_PPC_REG, ppc_last);
      debug_write(`DBG_CTRL_REG, 32'h0000_0001); // RESUME, but Single Step

zsr2018, 2019-11-28:
I want to know when into single step, the work manner with the core.

silabs-PaulZ, 2020-05-26:
Hi @zsr2018 , This bench is no longer supported and your question is not very clear. If you still have an issue can you clarify your question? Otherwise, can you please close this ticket?

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it either has been fixed with the stated (merged) pull request, because it is no longer valid, or because a follow up question for further clarification or feedback remained unanswered for at least 7 days. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.  Thank you for contributing to this core; raising issues will help to improve it.

closed 2020-07-14 by Silabs-ArjanB