openhwfoundation/cv32e40p #205: Parameter Reduction for CV32E40P  implementation

kenowhg, 2019-12-11:
The ri5cy core RTL parameters are included below.  There are many that will be removed we know, but from a verification point of view,  fewer is better,  zero is best.

For the CV32E40P implementation the number of parameters should be reduced as much as possible.  For sure parameters which cause the RTL to be different should be limited as much as possible.  Parameters which change the RTL make the collection of coverage difficult or impossible for verification efforts.

If parameters are required, the values that are legal should be indicated and limited.

module riscv_core
#(
  parameter N_EXT_PERF_COUNTERS =  0,
  parameter INSTR_RDATA_WIDTH   = 32,
  parameter PULP_SECURE         =  0,
  parameter N_PMP_ENTRIES       = 16,
  parameter USE_PMP             =  1, //if PULP_SECURE is 1, you can still not use the PMP
  parameter PULP_CLUSTER        =  1,
  parameter FPU                 =  0,
  parameter Zfinx               =  0,
  parameter FP_DIVSQRT          =  0,
  parameter SHARED_FP           =  0,
  parameter SHARED_DSP_MULT     =  0,
  parameter SHARED_INT_MULT     =  0,
  parameter SHARED_INT_DIV      =  0,
  parameter SHARED_FP_DIVSQRT   =  0,
  parameter WAPUTYPE            =  0,
  parameter APU_NARGS_CPU       =  3,
  parameter APU_WOP_CPU         =  6,
  parameter APU_NDSFLAGS_CPU    = 15,
  parameter APU_NUSFLAGS_CPU    =  5,
  parameter DM_HaltAddress      = 32'h1A110800
)

davideschiavone, 2020-01-28:
Issue #216 and #217 have been opened to do so.
APU* will be moved to a package and SHARED* removed as well as SHARED_FP.
This issue stays open for documentation

Silabs-ArjanB, 2020-05-19:
The parameter set has been greatly reduced. The corresponding documentation is now tracked via https://github.com/openhwgroup/core-v-docs/issues/45. 

@kenowhg If you agree that https://github.com/openhwgroup/core-v-docs/issues/45 serves the same purpose, could you please close this ticket?

MikeOpenHWGroup, 2020-05-19:
Hi.  @kenowhg is no longer on the project, so I will step-in.  The number of parameters has been greatly reduced and we are using good tracking to ensure that remaining parameters are properly documented.  I will close this issue.   Thanks to all.

closed 2020-05-19 by MikeOpenHWGroup