openhwfoundation/cv32e40p #206: Parameter documentation for CV32E40P

kenowhg, 2019-12-11:
The ri5cy parameters which are applied at the core top in the RTL, has limited or description in the provided documentation.  The parameters for the CV32E40P need to be fully documented in it's design spec.
Each parameter needs:
1)  Function, reason for being and justification.  (why do we need this parameter what does it do?)
2)  Details about how it will affect the RTL, if any.
3)  Document limits or valid values that can be used.

Silabs-ArjanB, 2020-05-19:
Replaced by https://github.com/openhwgroup/core-v-docs/issues/45

Silabs-ArjanB, 2020-06-19:
Hi @kenowhg , the parameters have now been documented in https://core-v-docs-verif-strat.readthedocs.io/projects/cv32e40p_um/en/latest/integration.html#parameters . Could you please check if this addresses your question and if so close this issue (or let us know what is missing)?

MikeOpenHWGroup, 2020-06-19:
Hi @Silabs-ArjanB.  When Ken raised this issue he was an employee of a BTA, an OpenHW member company.  He has moved on and is now working at Metrics.  However, even though Metrics is also a member company, he is not currently assigned to OpenHW tasks.

So I will take this one.  I will respond to it before the end of the today in Ottawa.

MikeOpenHWGroup, 2020-06-19:
Looks good @Silabs-ArjanB, closing.

closed 2020-06-19 by MikeOpenHWGroup