openhwfoundation/cv32e40p #207: ASIC synthesis target frequency

tangshibin, 2019-12-14:
What's the target frequency can be set for RI5CY (pulpissimo-v2.0.1) under tsmc40lup;
I saw it can be synthesized up to 650M Under 65nm;
But timing violation exists when it is synthesized for 200M under  40nm;
The timing violation is caused by a single-cycle 32x32 multiplier.
How can I improve the synthesis frequency?

HazemFathy, 2020-02-11:
Have you reached a solution for the performance limited by the multiplier? @tangshibin

barrydebruin, 2020-06-03:
I'm not sure if it helps, but i was able to synthesize it to 650MHz for 28nm RVT @ 0.9V, so it seems that there might be something wrong.

HazemFathy, 2020-06-03:
And how you reached that performanc? is it about specific timing constraints applied ?

barrydebruin, 2020-06-03:
No specific timing constraints. Im talking about the pulpino repo though.

HazemFathy, 2020-06-03:
Aha, thanks.

Silabs-ArjanB, 2020-06-19:
Hi @tangshibin It seems your question has been answered. Could you please close this issue?

closed 2020-06-20 by tangshibin