openhwfoundation/cv32e40p #208: Cross-contamination between RTL and TB

MikeOpenHWGroup, 2019-12-17:
I have noticed that some RTL source code is in the TB directory and that the associated variables in the Makefile at tb/core also mix RTL and TB files.

RTL source files in the TB directory:

1. fpnew is cloned into tb/core by the Makefile
2. tb/core/cluster_clock_gating.sv should be in rtl/cluster_clock_gating.sv

I would also recommend the TB* and RTL* variables in the Makefile be updated to ensure that TB* points only to files at or below tb/ and RTL* points to files at or below rtl/.  For example:
```
# TB source files
TBSRC_HOME  := ../../tb
TBSRC_PKG   := $(TBSRC_HOME)/tb_riscv/include/perturbation_defines.sv
TBSRC_TOP   := $(TBSRC_HOME)/core/tb_top.sv
TBSRC_CORE  := $(TBSRC_HOME)/core
TBSRC       := $(TBSRC_CORE)/tb_top.sv \
               $(TBSRC_CORE)/riscv_wrapper.sv \
               $(TBSRC_CORE)/mm_ram.sv \
               $(TBSRC_CORE)/dp_ram.sv \
               $(TBSRC_HOME)/tb_riscv/riscv_random_stall.sv \
               $(TBSRC_HOME)/tb_riscv/riscv_random_interrupt_generator.sv
TBSRC_VERI  := $(TBSRC_CORE)/tb_top_verilator.sv \
               $(TBSRC_CORE)/riscv_wrapper.sv \
               $(TBSRC_CORE)/mm_ram.sv \
               $(TBSRC_CORE)/dp_ram.sv

# RTL source files
RTLSRC_HOME   := ../../rtl  # will be ../../../rtl for CV32E
RTLSRC_INCDIR := $(RTLSRC_HOME)/include
FPNEW_PKG     := $(RTLSRC_HOME)/fpnew
RTLSRC_PKG    := $(FPNEW_PKG)/src/fpnew_pkg.sv \
                 $(addprefix $(RTLSRC_HOME)/include/,\
                   apu_core_package.sv \
                   riscv_defines.sv \
                   riscv_tracer_defines.sv)
RTLSRC        := $(filter-out $(RTLSRC_HOME)/riscv_register_file_latch.sv,\
                   $(wildcard $(RTLSRC_HOME)/*.sv))
```

I have a local branch with these and other changes, but I can no longer push a branch to this repo.

bluewww, 2019-12-18:
`tb/core/cluster_clock_gating.sv` is mean to be a testbench file, it is not a part of the RTL. This is why it is also not exported in https://github.com/pulp-platform/riscv/blob/master/src_files.yml. I can change the name to make this more clear.

As for your other suggestion I think they are useful. I'll prepare a PR.

MikeOpenHWGroup, 2019-12-18:
Thanks for looking at this so quickly.   I'm pretty sure that cluster_clock_gating.sv is used by the RTL.
On a related note - would it be possible to reinstate my permissions so that I can push a branch to this repo?

bluewww, 2019-12-19:
I just checked I don't see any branches of yours. Your last PR made was with a fork. @davideschiavone

MikeOpenHWGroup, 2019-12-19:
That correct, I do not have the privileges to create a branch on this project.  I _thought_ I had these privileges in the past, but was mistaken.  As you say, my previous PR was from a fork.  I can create another branch on my forked project if you like.

MikeOpenHWGroup, 2020-02-11:
Hi.  I'd like to resurrect this issue now that the repo has moved.  The cv32e40p RTL instantiates `cluster_clock_gating` in [riscv_core.sv](https://github.com/openhwgroup/cv32e40p/blob/master/rtl/riscv_core.sv) starting at line 458.
[image]
There is nothing wrong with this except that the module is defined in the testbench at [tb/core/cluster_clock_gating.sv](https://github.com/openhwgroup/cv32e40p/blob/master/tb/core/cluster_clock_gating.sv).  It is also defined in the [tb/verilator-model](https://github.com/openhwgroup/cv32e40p/blob/master/tb/verilator-model/cluster_clock_gating.sv).   The two files are identical.

Neither the `core` testbench nor the `verilator-model` testbench use this module.  The RTL clearly does, so the file should be in the RTL directory.

Generally speaking verification people should not make updates to the RTL, but I can make an update and issue a pull-request if you like.  If you prefer make the change yourself.

Thanks!

davideschiavone, 2020-02-12:
So basically, the cluster_clock_gating is a simulation-only module.
That's why @bluewww called it "a testbench file", probably he meant that it's not synthesizable (I know... we designers call testbench everything that is not real hardware :))

I opened a new issue to create such simulation only file in the RTL folder of the core:
https://github.com/openhwgroup/cv32e40p/issues/228

MikeOpenHWGroup, 2020-02-12:
Understood.

So is this issue to be closed?   This issue is gating verification as we currently cannot compile.

davideschiavone, 2020-02-12:
Why can't you compile? Is it because you are not compiling the file in the tb folder?
If so I we can move it to the RTL and give another name to it (_cluster_ comes from PULP, I suggest now `_core_clock_gating_` ).

MikeOpenHWGroup, 2020-02-12:
The Makefiles in core-v-verif have been modified to have a strict division between the RTL, TB, testcases and simulation results.   Also, anything that should be in an RTL directory has been removed from the TB directory.

MikeOpenHWGroup, 2020-02-13:
Any progress on this?  I need this one resolved to make progress.   I am OK with Davide's suggestion, but won't do it myself (I want to keep the verifiers out of the RTL).

davideschiavone, 2020-02-13:
We have a PR pending for reviews
https://github.com/openhwgroup/cv32e40p/pull/232

Silabs-ArjanB, 2020-06-25:
Hi @MikeOpenHWGroup Can this issue be closed now?

MikeOpenHWGroup, 2020-06-25:
Yes it can.   Thanks.

closed 2020-06-25 by MikeOpenHWGroup