openhwfoundation/cv32e40p #209: User Manual: Errors in Hardware-Loop CSR Mapping table

MikeOpenHWGroup, 2019-12-22:
The RI5CY User Manual (https://github.com/pulp-platform/riscv/blob/master/doc/user_manual.doc) has several copy-n-paste errors in **Table 6: Hardware-Lopp CSR Mapping**.   Specifically:
* Address bits [5:0] for lpstart[1], lpend[1] and lpcount[1]
* Description of lpstart[1]

closed 2020-01-28 by davideschiavone

MikeOpenHWGroup, 2020-01-28:
Hey Davide, when you close an Issue, what does it mean?   Did you fixed it, or was there no problem to begin with and  you are not going to change anything?   If you made a change, where is it?

bluewww, 2020-01-28:
He wrote #fix 209 in the commit which is linked which automatically closed this issue

MikeOpenHWGroup, 2020-01-28:
Thanks bluewww (hopefully I'll use your _real_ name one day).  Is this the typical process for closing issues?

bluewww, 2020-01-28:
Normally you make a pullrequest that explains how and what is being fixed. In the pullrequest you can then somewhere write "fixed #issuenumber" which links issue and PR together. You would then get a notification in this thread that a solution is in this PR. The solution can them be discussed and reviewed there and finally merged closing the linked issue automatically.

Devs sometimes directly commit and push to master which is what happened here. It is not recommended in my opinion because it sidesteps the discussion-of-the-fix phase but practically people dont comment on merges very often and we also discuss some fixes internally.

To *enforce* the PR way we could make the master write protected. It is also possible to make reviews mandatory (see github review function). This is how lowrisc does it.

davideschiavone, 2020-01-29:
As Robert says (this is his name :)) there should be a PR and a review process.
This is what lowRISC and OpemHW will do.
In this case, the documentation is not in text format yet (so difficult to review changes), it is not involving code, plus I am the only reviewer at the moment :) So I made I short cut :)

commit 98677759 (2020-01-28): New HWLoop constraints and fix #209