openhwfoundation/cv32e40p #214: RI5CY: User Manual v4.2: p.clipr instructions opcode seems wrong

alfredoh1234, 2020-01-23:
RI5CY: User Manual v4.2 section 14.3.3 appears to show inaccurate p.clipr opcode:

| **Mnemonic** | **Description** |
| -- | --|
| p.clipr rD, rs1, rs2 | if rs1 <= **-(rs2+1)**, rD   = **-(rs2+1)**, <br /> else if rs1 >=rs2, rD = rs2, <br /> else rD = rs1                                 |

Is it indicating that the value stored in rs2 should be inverted before checking if rs1 <= -(rs2+1)

Shouldn't the opcode show rs2 to be signed and the opcode be something like rs1 <= Sext(rs2+1) , rD = Sext(rs2 +1)

davideschiavone, 2020-01-28:
If not specified, numbers are always signed (as in RISC-V, where they specify when numbers are not signed).
Sometimes the Sext is added when the number is not 32bits in the spec, to specify there is a sign extensions. But rs2 is a register, thus 32bits

alfredoh1234, 2020-01-30:
Can there be a sentence added to the document saying that? For example between the header of section 14 and the first subsection 14.1 ?
Something like: "For numbers that are not 32-bit, sign-extension (Sext) is specified; but this is not the case for General Purpose Registers (GPRs)." like rs2 in this case