openhwfoundation/cv32e40p #215: RI5CY: User Manual v4.2: ALU Ext. (Ch 5, 14.3.3, 14.3.4) needs a bit more details

alfredoh1234, 2020-01-23:
p.clip*, p.add* and p.sub* instructions do not indicate that operands are Q.number-format.

Clarification (or URL) would be welcome to have all information needed in this document for implementation, debug and verification.

davideschiavone, 2020-01-28:
Hi @alfredoh1234 ,

why the Q format is important for these spec? It should be independent on the format. If you see something that has implication on the spec, then the spec should be rewritten as the numbers are simply treated as 32bits number. The Fixed Point idea is just in the programmer mind :) but not in datapath (differently from floats)

alfredoh1234, 2020-01-30:
Ok. I will close the issue based on this information.

closed 2020-01-30 by alfredoh1234