openhwfoundation/cv32e40p #217: remove deprecated parameters

davideschiavone, 2020-01-24:
Remove and clean the code for:

```
  parameter FP_DIVSQRT          =  0,
  parameter SHARED_DSP_MULT     =  0,
  parameter SHARED_INT_MULT     =  0,
  parameter SHARED_INT_DIV      =  0,
  parameter SHARED_FP_DIVSQRT   =  0,

```
Make **parameter SHARED_FP**  always equal to 1

davideschiavone, 2020-01-27:
According to @stmach , SHARED_FP is not a binary parameter.
The maintained version is SHARED_FP equal to 2.

Here the commit where the FPU is always outside the core. The SHARED* parameters have been removed.

2 pending questions:

What should be the value of SHARED_FP_DIVSQRT ? 
1 or 2?

Should the ALU still contain FPU instructions?

Silabs-ArjanB, 2020-10-28:
Done

closed 2020-10-28 by Silabs-ArjanB