openhwfoundation/cv32e40p #218: Define "Retire" signal to indicate an instruction has completed

GTumbush, 2020-01-24:
I'm using the Imperas ISS as the golden model to verify a RI5CY implementation.  As seen below the ISS has a “Retire” signal indicating an instruction has completed and I can capture the ISS’s current state (PC, GPR, etc). This would be used to compare against the RI5CY RTL internal state.  I'm requesting a similar “Retire” signal from the RI5CY RTL to indicate the same.  This signal can be internal to the core and can be a combination of signals.

[image]

Davide Schiavone asked me to open this issue based on a Jan 23/24, 2020 email conversation.

eroom1966, 2020-01-29:
@GTumbush @davideschiavone 
Having taken a (brief) look at the RTL, I would guess that a similar decoding must be done within the riscv_tracer.sv file. I am assuming this is producing a disassembly of executed instructions, in which case it must be able to decode when an instruction has completed and hence - retired.
I think all we would need to do is to enhance this in order to provide an indication of the address of the instruction which has been retired at this completion stage. 
I would delve down into this module, but it looks pretty complicated, and I am sure the author will know exactly how to indicate the address of the retired instruction and generate an event or status signal :wink: 

thx
Lee

davideschiavone, 2020-01-31:
Yes that's correct. The simulation tracer does this job quite decently.
But there is no "single" signal. Basically the tracer replicates the pipeline flow in a simulation way and then it prints the instruction in the log. To know where the status is, you can see where the instruction is actually printed in the log. There you have all the info :)

davideschiavone, 2020-02-24:
487c8ac627e6fc7ead14c998a9a2f613e2ac57fe

closed 2020-02-24 by davideschiavone