openhwfoundation/cv32e40p #221: SCM-based register file is not cleared at reset

FrancescoConti, 2020-01-31:
This causes RTL vs gate-level simulation mismatch. @davideschiavone I can submit a PR with a fix for this (it's the same I already implemented e.g. in https://github.com/pulp-platform/hwpe-ctrl ). Would you consider merging it?

davideschiavone, 2020-01-31:
really ? I never had problem with post layout simulations. What kind of problem is that ? can you describe it a bit more? of course I consider your PR to be merged :)

FrancescoConti, 2020-01-31:
It's not exactly a bug, but in some cases to make PS simulations work I had to manually clear the register file in Modelsim with `force -deposit regfile/Q 1'h0`. It's a bit annoying :)
Right now I'm doing that on a synthesis of a somewhat recent version of PULPissimo, the testbench dies miserably when the debug module tries to halt the core.

bluewww, 2020-02-04:
But doesn't that mean it's an issue with the testbench? I think its a quite common assumption that cpu registers are unitialized/hold random garbage on reset.

FrancescoConti, 2020-02-04:
@bluewww this is *mostly* a testbench related issue . But actually if you implement the register-file with flip-flops you will get a subtly different state, when you assume that the state is identical, leading to divergence with expected behavior in case of buggy code (I actually saw this on FPGA with a much older version of Riscy :) )

gautschimi, 2020-02-04:
you should clear all relevant registers in the boot sequence. or better write code that does not depend on uninitialized registers:)

davideschiavone, 2020-02-04:
Yes, in my opinion this comes from buggy code - as you should not use a register content if not utilized.

However, as @FrancescoConti wrote in the PR, resettable latches are more expensive and not always available in libraries. The cost of this change looks negligible (1FF and 33 OR2 gates).

 I do not have a strong opinion neither to accept nor to deny this PR - I can actually forwards the question to openHW

FrancescoConti, 2020-02-04:
Absolutely, no strong opinion on my side either -- just a proposal as I think this is the kind of subtle difference that it's better to avoid at the root.

davideschiavone, 2020-02-04:
What I like about accepting it is that the 2 cores (FF or Latches) would have the same state.
Also it would save some instructions.
The cons is that actually it should be a SW task.

MikeOpenHWGroup, 2020-02-04:
In industry it is common practise to reset everything, unless it becomes impractical in physical design (all those resets create a lot of loads).  The idea is to protect software from itself.

Having said that, the final word should be with the ISA.  I seem to recall that the ISA explicitly states that the register file is not initialized by hardware (except for x0).  Alas, I can no longer find it.

If anyone can point this out in the ISA, then we have our answer.  Otherwise, I have a mild preference to reset everything, but no strong argument for it.

MikeOpenHWGroup, 2020-02-04:
I had a question in an off-line thread about my previous comment on this issue, so I'll post my reply here.

Resetting the registers is considered "safer" because software doesn't always do what it should.  Stating that s/w should never use an uninitialized register means that you are right, but the product still doesn't work.  Finding the problem in the lab is hard.  Living with the problem in the field is _really_ hard and rather expensive.  So often things are done in hardware to protect software from itself.

Again, the ISA should have the final word on this.   If it states that registers are not initialized by h/w, then we should not do it.

bluewww, 2020-02-04:
I have this from `riscv-privileged-v1.10`, `section 3.3 Reset`:
```
Upon reset, a hart’s privilege mode is set to M. The mstatus fields MIE and MPRV are reset to
0. The pc is set to an implementation-defined reset vector. The mcause register is set to a value
indicating the cause of the reset. *All other hart state is undefined.*
```

davideschiavone, 2020-02-04:
This means that it can also be 0s. You now I am more convinced to accept the PR.
Waiting for more people in OHW to see where it goes, but so far we have:
PRO:    SW leverages this
CONS: A bit of overhead

This bit of overhead is little enough to justify the easier life the SW would have in the future.

Saying that the FF and the Latch based RF are different is not an option for accepting or denying the PR, as whatever we do, we should align the other RF as well. So I think that if we deny this PR, we then remove the reset from the flip-flop based RF as well.

davideschiavone, 2020-02-04:
Ah by the way, Ariane resets the RF as well. So we should align Ariane as well whatever we decides now.
In addition, Ibex from LowRISC has the same issue in the Latch RF, so pinging @vogelpi to bring this up in LowRISC as well

gurkaynak, 2020-02-04:
Hello, 

First off I think all state holding elements (especially RF) should have a proper reset. I do not see the point of having async reset for FF and not for Latch based implementations of the same block, and I would advise against implementations that provide a work around for the case that there might be some technologies for which the std cell library does not contain by default resetable Latches. This is an exception, and if someone needs to implement it for this technology they are welcome to add a specific solution but the RF should have standard asynchronous reset. 

I am very skeptical of the 'gains' that a Latch based RF has over a FF based one, especially if in an industrial design you need to add DfT solutions to it. I would argue the overhead needed for the DfT solution would take away most (if not all) the advantages of using the latch based version. I would still keep the latch based solution as an alternative but concentrate the efforts of the FF based one.  

So in short: 
- no to work around
- add proper reset to the latch based solution
- concentrate on the FF based solution

I think part of the discussion comes from the fact that some wanted to re-use the SCM (which was designed to replace SRAM cuts) as RF, but I might be mistaken.

davideschiavone, 2020-02-04:
So @gurkaynak made a good point about the work around to do the reset. 
The usual
`if (~reset) else `
statement should go into the always_latch process without the workaround.
Same regarding the use of the FF for a product that indeed needs DFT. 

Regarding the "RF must have a reset" as we said, it makes life easier but it is not a spec.
However, it seems the right way to go, with all the cores and their configurations aligned.

So I would say, let's make the reset in the always_latch process without the workaround.
If you all agree thumb up :+1:

FrancescoConti, 2020-02-05:
@davideschiavone I aligned the PR to the reset directly in the `always_latch`, waiting for CI result (but I already did a few tests locally, no surprises there).

davideschiavone, 2020-02-05:
Thanks a lot @FrancescoConti .

This fruitful discussion ended up having now the register file with flip flop aligned with the one with latches, and aligned to Ariane. ( Ibex is still not aligned @vogelpi )

This was not mandatory by spec, but highly recommended to make software life easier.

Thanks everyone for your contribution to the topic

Davide

Issue solved at 90428d403ee06a8de0ad9571cc3d0b9ba51ee674

closed 2020-02-05 by davideschiavone

vogelpi, 2020-02-05:
Thanks for pinging me @davideschiavone . Indeed latch-based register file in Ibex has the same issue.

Like @gurkaynak I also think that state-holding elements should have a proper reset.

But I consider it problematic if this is done to protect software from itself and especially if software relies on it. As mentioned there are always cases/technologies where resets are available. One is example are std cell libraries not providing latches with resets, another one are FPGAs where the use of LUTRAM (no reset) is highly preferred for register files due to brutal savings in terms of resource utilization. For example, in Ibex this allows to implement the whole register file with 48 LUTRAMs instead of around 600 LUTs + 1000 FFs. I think these reasons are why the spec does not demand a defined reset state for the reg file, and why software should not rely on the hardware doing the right thing.

zarubaf, 2020-02-05:
The same holds true for high-performance CPUs where registers files are larger than the architectural visible ones because of re-naming. These will likely be implemented out of SRAMs which usually don't provide any reset. Furthermore, I don't really understand why zero should be a legal value for software to assume. It is as wrong as any other random value which will be in the register file.

Unfortunately, this wrong X propagation is an issue in Verilog (imho). The reset routing was a serious problem in the larger (and smaller tech) chips we were doing recently. This is why I am advocating for registers that don't need to be reset to don't have one.  What we (and others) do for post-layout simulation is to dump all registers from the netlist and do the `deposit` magic on them before starting the simulation. 

Just my two cents on this topic, which is anyway already resolved :-D

arjanbink, 2020-02-06:
I also am strongly in favor of having a proper reset for all latches/flops.

davideschiavone, 2020-02-06:
Thanks @arjanbink , indeed this is now part of the core