openhwfoundation/cv32e40p #225: the long path of multiplier

omarabelr, 2020-02-09:
I design the physical design of this core and I face a problem, the multiplier has a long path this limit my speed to 25 MHZ in the worst case in my technology.
what is your advice for me to solve this problem to keep the same logic.

davideschiavone, 2020-02-09:
dear @omarabelr ,

can I ask you which technology you are using and what are you synthesizing ? can you describe the system around the core ?
thanks

omarabelr, 2020-02-09:
dear @davideschiavone ,

I use (saed32hvt_ss0p7v125c) technology and I synthesis the design on design compiler(Synopsys synthesis software products) and my project is to do ASIC physical design for this core.
And after synthesis I see that there are many paths related to the multiplier that have a long combinational logic.
thanks.

davideschiavone, 2020-02-10:
Usually the critical path (CP) is between the core and the memory cuts.
25MHz seems very slow in 32nm. However, with the few details I have the only thing I can suggest to speed up the multiplier is not implement a 2 cycles version of it (you can get inspired by the MULH to build an FSM that does multiplications in a multicycle fashions)

best
Davide

HazemFathy, 2020-02-11:
Could you please specify the exact multicycle paths inside the core? @davideschiavone 

Thanks in advance.

davideschiavone, 2020-02-11:
It's not a muticycle path, it's an FSM (iterative implementation if you prefer).
Here you see the FSM that implements the MULH
https://github.com/pulp-platform/riscv/blob/master/rtl/riscv_mult.sv#L132

You should be something similar for the multipliers that are too long for your timing constraints

HazemFathy, 2020-02-11:
we're using this implementation in synthesis of the core as well as alu_div.sv with FSM implementation also.
when we're viewing timing reports, we see long timing paths starting from the operands' registers in (riscv_id_stage.sv) passing by "result_o" the output port of the multiplier crossing to the id_stage again and passing the result for register_file or continuing to the operands themselves again.

also, i'm setting case analysis for signals "operator_i" in multiplier to choose "MUL_H" and for the divider to choose division operation.

attached two log files reporting the violations that are mostly related to the multiplier and divider and a timing report for a certain path related to the multiplier as an example.

[max_timing.log](https://github.com/pulp-platform/riscv/files/4185960/max_timing.log)

[violators.log](https://github.com/pulp-platform/riscv/files/4185961/violators.log)

davideschiavone, 2020-02-11:
There was a typo in my answer sorry. You critical path goes through the multiplier, so you have to change the design and implement a new multiplier that takes more than 1 cycle to break the long path. To design it, I suggest you to get inspired from the MULH multiplier that I provided the link above. 

You have to do this for all the data path components that are too critical for your constraints.

HazemFathy, 2020-02-11:
i'm already using the multiplier you've provided above and these violations appear with it, you mean there's something wrong with it and I should design my own one?

davideschiavone, 2020-02-11:
Yes, you should design your own.
The critical path seems to be in the dot product unit, so probably you have to design your own. I am suggesting a 2 cycles implementation. 

I'e never seen the divider in our critical path. I fear it's not really there, but you may want to add your divider long path

HazemFathy, 2020-02-11:
but I found in the user manual that RI5CY is capable of executing 32x32 multiplications in one cycle and only upper-half word results in 4 cycles , and division operations are executed between 2 to 32 cycles.

does that mean that multicycle multiplication and multicycle division are somehow pipelined or what ?

davideschiavone, 2020-02-12:
Yes, the RTL is designed for a one cycle 32x32 multiplication. If the design does not match your constraints, you have to change the design to make it as fast as you need

HazemFathy, 2020-02-12:
if i registered the outputs of the multiplier and ALU , that won't affect the functionality of the core , right ? 

i see they're mostly contributing to the TNS and WNS in the design.

davideschiavone, 2020-02-12:
Yes it changes the functionality. You have to play with the _ready_ signals in the alu and multiplier

HazemFathy, 2020-02-12:
Aha, thanks for your help.

Silabs-ArjanB, 2020-06-15:
Hi @HazemFathy Your question seems to have been answered. If you agree could you please close this ticket? (The ticket will get closed automatically in 7 days in case no further follow up is received)

Silabs-ArjanB, 2020-07-14:
We are closing this issue as it has been answered at least 7 days ago. Please do not hesitate to re-open this issue if you do not agree that the issue has been handled satisfactorily or if you have further questions.

closed 2020-07-14 by Silabs-ArjanB