openhwfoundation/cv32e40p #228: Creating a core clock gating module in this repository

davideschiavone, 2020-02-11:
Currently, the core does not have in its own repository the cluster_clock_gating cell used to clock gate the core on WFI instructions.

Under the PULP platform simulation, this module is available in another repository, and it can be used only for Simulation.

In Silicon or FPGA implementations, the cell should be replaced to the technology one.

Making this module part of the core, means:

- either forking the core, change the module with the technology related cell, and push it back to a private channel every time a tapeout or an FPGA implementation is done

- or being sure that compile and synthesis tools link the technology cell at simulation and synthesis time.

Both the options will be available, whereas the open-source repository will maintain the generic simulation clock gating cell.

MikeOpenHWGroup, 2020-02-12:
The second option (compile and synthesis tools link the technology cell at simulation and synthesis time) is by far the preferred option.

There should be one and only one set of manifest used by everyone.  The manifest is where the intelligence to determine which files are used for synthesis and which for simulation is located.

davideschiavone, 2020-02-12:
Yes, anyway these options will be always available to all the users :) they can do whatever they want. But I am also in favor for 2.

MikeOpenHWGroup, 2020-02-12:
We can't control how people use our work, but we _can_ control how _we_ use it.

closed 2020-05-17 by davideschiavone