openhwfoundation/cv32e40p #229: Should we remove the TB directory from this project?

MikeOpenHWGroup, 2020-02-11:
The OpenHW Group CORE-V projects have separate GitHub projects for design and verification.  As such, the production testbench for cv32e40p is not the one in the `tb` directory.  Given that, I recommend it be removed from this repo altogether and replaced with a pointer to the core-v-verif project.

The obvious objections to this proposal is that the testbench in cv32e40p can be used by the designers for basic sanity.   This is not a good idea because it allows design updates to be made without ensuring the testbenches are kept up to date.  Basic sanity means the RTL and TB are compatible and work together.  This must always be true.  So if a designer needs to run a sanity test (which will always be supported), they need the core-v-verif project.

davideschiavone, 2020-02-12:
I would keep the tb folder in there as many people may want to use the core with a minimal setup to run a `hello world` simple  C program without cloning other repositories.
Rather than removing such folder, I would simply change its name and meaning.
It's correct that this means maintaining that folder too, but it's so simple (and we can even make it simpler) that it's a good feature to have.

A name could be something that tells the user that in there you can run simple program. Any idea?

@bluewww , @MikeOpenHWGroup

bluewww, 2020-02-12:
I'm also for keeping the folder but the meaning should be downgraded from `testbench` to something like `example usage`.

davideschiavone, 2020-02-12:
I like `example`, but it doesn't explain a lot.
How about `sim` ?

MikeOpenHWGroup, 2020-02-12:
In core-v-verif we use "sim" to run tests.  The only committed file in "sim" is the Makefile - this keeps all simulation generated files in one place.

If you _must_ keep it, you could call it `example_tb`.   But I **_strongly_** recommend against it.

MikeOpenHWGroup, 2020-02-18:
Hey everyone, we should not let this issue get "stale".  I am already getting questions about the 'redundant" testbench in this repo, so we either need to get rid of it, or rename it to be clear that its not a production testbench.

davideschiavone, 2020-02-19:
I would still call it "sim" as it is actually its meaning. It runs tests/programs.
I propose otherwise `exe` or  `test_program`.  
Let's pick one and move forwards

MikeOpenHWGroup, 2020-02-19:
I'm not in favor of this.  The cv32e40p cannot be considered "sane" if it
doesn't work in its proper test bench.   So keeping a test bench in that
repo is a bad idea, regardless what we call it.

---mike

davideschiavone, 2020-02-19:
It's not a testbench anymore. I agree it shouldn't have a testbench in this repo.
But it should have an environment where an user can run `Hello world`. Mostly every core has this in its repos

Silabs-ArjanB, 2020-04-07:
I would prefer if at least the following are removed from the cv32e40p repos (to avoid the code duplication and related confusion on where to find the actual testbench):

Files:

cv32e40p/tb/dm/tb_top_verilator.sv
tb/core/tb_top_verilator.sv

Directories:

cv32e40p/tb/verilator-model
cv32e40p/tb/serDiv
cv32e40p/tb/tb_MPU
cv32e40p/tb/tb_riscv

Ideally also all of the compliance tests and debug (dm) test case / test bench are removed (assuming that integration with the actual debug module will be supported in core-v-verif). The only test case we could keep here would be 'hello world', but even that will lead to code duplication between cv32e40p and core-v-verif, so I would definitely not be against only having RTL code in the cv32e40p repos.

Silabs-ArjanB, 2020-06-23:
Hi @davideschiavone , @MikeOpenHWGroup Can we now remove the above stated files and directories and also cv32e40p/tb/dm?

MikeOpenHWGroup, 2020-06-23:
Yes from my end.

Silabs-ArjanB, 2020-07-14:
Hi @MikeOpenHWGroup Most of the testbenches and test cases have been removed from the cv32e40p repos now. We kept 3 examples: 'hello world', 'floatinf point example,' and 'interrupt example'. If you agree with this can you please close this issue; if not, please let us know what more needs to be done.

MikeOpenHWGroup, 2020-07-14:
Thanks @Silabs-ArjanB.  I have noticed over the past few weeks that these changes have cleared up a lot of confusion.  Much appreciated.  I will close this issue.

I've requested a small README be added to the example_tb directory, see https://github.com/openhwgroup/cv32e40p/pull/409.   That PR need not gate closing this issue.

closed 2020-07-14 by MikeOpenHWGroup