openhwfoundation/cv32e40p #230: How to handle new interrupt inputs

MikeOpenHWGroup, 2020-02-11:
The CV32E40P has a number of new interrupt inputs:
[image]
Is it safe to tie all these off to 'b0 until such time as the testbenches can do something interesting with them?

Related question: when/where will the function of these new inputs be documented?

davideschiavone, 2020-02-12:
Yes it's safe.
These inputs will be documented directly in the new document, for the time being one can use the Ibex documentation (these changes come from there) for all the interrupts but irq_fastx_i (which are documented nowhere for the moment)

davideschiavone, 2020-02-12:
I would keep it open as a reminder that the Interrupt should be documented

MikeOpenHWGroup, 2020-02-12:
Then let's re-label it as a `task`.