openhwfoundation/cv32e40p #243: Why are only 5 bits of the IRQ id used for irq_id_o and exc_cause_o?

Silabs-ArjanB, 2020-02-26:
I see in https://github.com/openhwgroup/cv32e40p/blob/master/rtl/riscv_controller.sv that only the least significant 5 bits of the IRQ identifier are used when assigning irq_id_o and exc_cause_o. Furthermore irq_ack_o is only asserted when irq_id_ctrl_i[5] and not for the other interrupts.

Could you explain why not all interrupt sources are being treated equally?

davideschiavone, 2020-02-26:
Because we though to use the custom id_o+ack protocol only for the CLINTx extensions to avoid confusion for those who want to use only the CLINT interrupts (the Ibex ones).
In this way, systems that use the old PULP-way to interact with interrupts do not have to change the interrupt controller (as this is still 5bits), but only to add a one-hot-encoder to connect the old irq + id to the 32 fastx interrupt lines.

The systems that use the Ibex way, can leave unconnected(grounded) the fastx interrupts and use the CLINT interrupt lines. The ACK in this case is not used to avoid confusion (internal decision at ETH but can be changed if needed)

The mcause instead has been extended otherwise software that uses 64 interrupts can't work.

The exc_cause_o is used to calculate the offset of the IVT to jump to. As there are 2 different IVTs (one pointed by MTVEC and one by MTVECx) , the exc_cause is only 5 bits. If the interrupt taken is a FASTx, the MTVECx is selected, otherwise the MTVEC.

Silabs-ArjanB, 2020-02-27:
@davideschiavone Could you please point me to the intended interrupt controller for both CLINT and CLINTx interrupts? What if somebody wants to use all the interrupt inputs; will they need 2 interrupt controllers or will they somehow need to differentiate in their software on how they clear interrupts based on their id?

davideschiavone, 2020-03-10:
Hello @Silabs-ArjanB , there is none for the moment. It's a feature that has no interrupt controller designed for it. We simply disabled the ACK signal from the core for the CLINT to avoid confusion when people want to use it without (personally I don't like it though).
If we want to change it I am complety ok with it.
What do you think?

PR #358 (2020-06-09): IRQ related fixes (removed NMI, removed MTVECX, unified fast/fastx interrupts)
IRQ related fixes as documented in https://github.com/openhwgroup/core-v-docs/issues/50. 

- Removed NMI
- Unified fastx and fast interrupts
- Removed MTVECX
- Made irq_ack_o and irq_id_o apply to all interrupts

Fixes #243 #323 #336
closed 2020-06-09 by davideschiavone