openhwfoundation/cv32e40p #244: interrupt_test.hex produce "illegal instruction handler entered" message

shirley85159, 2020-02-27:
Hi there,
I was trying to run tests for the core using vsim, firmware and custom work fine for me but I got following error when I try to run interrupt_test. I used make interrupt-vsim-run for it. Is there any reason this happens? Thanks a lot!
[image]

davideschiavone, 2020-04-07:
Hello, I cannot reproduce this error. Can you give us more insights? maybe the objdump of that portion of code? Did you look at the trace?

MikeOpenHWGroup, 2020-04-07:
Hi @shirley85159.  When reporting a bug, it is helpful to assign the issue to specific individual (I usually assign issues to @davideschiavone).  It is also import to include the following:

### Steps to Reproduce
1. URL to branch (and maybe the hash) that exhibits the issue.
2. Command line  (e.g. `make xrun-riscv-compliance`)
3. Logfile and/or wave-dump info (screen shots can be useful)

You have already provided item 3.

Silabs-ArjanB, 2020-06-15:
Hi @shirley85159 Could you please check if there is still an issue here (for me the stated test runs fine on today's master branch (748abbf0bfdf3fd0f3e9335c3aefa93d25fe88d0)? Please provide means for us to recreate the issue. (This issue will be close in 7 days if no further input is received.)

closed 2020-06-24 by davideschiavone