openhwfoundation/cv32e40p #252: Align the code to the lowRISC style-guide

davideschiavone, 2020-03-05:
The code should be restructerd following the lowRISC style-guides defined at

https://github.com/lowRISC/style-guides

Smattacus, 2020-03-25:
Is this still open? I'd like to start getting involved, and refactoring code is a way I can learn what's going on internally.

davideschiavone, 2020-03-30:
Hello @Smattacus !

Yes that would be amazing! Would you like to set a call up with me and @rickoco ?

It would take less than 30 minutes.

Thanks
Davide

Smattacus, 2020-03-30:
Great! What calling software do you use? Wednesday and on is best for me. I am in USA PST.

MikeOpenHWGroup, 2024-06-04:
This was not resolved for v1.0.0 and we are now at v1.8.0.  As we approach v2.0.0 we should consider whether to:

1. Close it.
2. Waive it for v1.8.0 (and by extension v2.0.0).

Making existing code conform to a new coding style is a nice-to-have, not a must-have feature, so I recommend option 2.

pascalgouedo, 2024-06-04:
+1 for waive it.

davideschiavone, 2024-06-04:
waive