openhwfoundation/cv32e40p #254: mtvec is tied to boot_addr_i when PULP_SECURE=0

jm4rtin, 2020-03-09:
When user mode is not added to the core (PULP_SECURE=0) the mtvec CSR has a fixed value (boot_addr_i) that can't be changed. We would like the ability to change this address even if user mode is not enabled in the core.

davideschiavone, 2020-03-10:
Dear all, @Silabs-ArjanB , @sjthales , @haugoug   what do you think about it?
Having the MTVEC read-only is ok by SPEC and it gives the possibility to save one 32b register.
I would like to ear your opinion to decide whether we want to have it READ-ONLY like today or writable like @jm4rtin  proposes.

bluewww, 2020-03-10:
I think for `PULP_SECURE=1` the `mtvec` register is 24 bits (lower 8 bits zeroed). In my opinion it makes sense to also make `mtvec` dynamic for `PULP_SECURE=0`.

There is obviously always the last possibility of making this behavior (hardcoded mtvec vs dynamic mtvec) configurable with a parameter, though I'm not sure how happy the dv team is about another parameter.

Silabs-ArjanB, 2020-03-23:
I would definitely prefer to make it possible that mtvec differs from boot_addr_i when PULP_SECURE=0. I would be fine with either adding a new input signal (mtvec_addr_i) or making mtvec(x) an actual register (same as PULP_SECURE=1. I have a preference for making this identical to the PULP_SECURE=1 behavior.

davideschiavone, 2020-03-23:
It makes sense to me that we do as PULP_SECURE=1. 
@jm4rtin would you like to prepare a PULL REQUEST? @Silabs-ArjanB and I will review it gladly

davideschiavone, 2020-04-02:
https://github.com/openhwgroup/cv32e40p/pull/285

closed 2020-04-02 by davideschiavone