openhwfoundation/cv32e40p #256: Retire signal not correct for CSRs

GTumbush, 2020-03-10:
Opening this issue based on call March 10, 2020. 

The tracer does not support printing CSR's to the tracer file.  I attempted to compare cv32e40p mstatus_q to mstatus predicted by the Imperas ISS and experienced inconsistent timing of mstatus_q update for the mret and ecall instructions. 

The mret instruction updates mstatus_q to 8, 1 clock cycle after the retire signal (see below).
[image]

The ecall instruction updates mstatus_q to 6, 2 clock cycles before the retire signal (see below).
[image]

The software to produce the above is in the attached zip file.
[riscv_arithmetic_basic_test_0.zip](https://github.com/openhwgroup/cv32e40p/files/4313734/riscv_arithmetic_basic_test_0.zip)

GTumbush, 2020-03-10:
Adding the cycles signal to make the mret and ecall instructions easier to find:
[image]

[image]

GTumbush, 2020-03-18:
wait(retire_delay) in riscv_tracer.sv does not compile with Incisive 15.20.077 or Xcelium 19.09.007

xmvlog(64): 19.09-s007: (c) Copyright 1995-2019 Cadence Design Systems, Inc.
      wait(retire_delay);
                      |
xmvlog: *E,EVNOTR (/wrk/gtumbush/mcu/riscy/cv32e40p_fix256/riscv/rtl/riscv_tracer.sv,854|22): An event is not a legal rvalue in this context [9.7.3(IEEE)].

Change to @(retire_delay);

GTumbush, 2020-03-18:
This fixes the issue with mret but not ecall.  

This is the new waveform for mret. Looks good!
[image]

Waveform for ecall is the same

GTumbush, 2020-03-18:
Note that the mepc register is still miscomparing for the ecall instruction in the same way as the mstatus.  It updates 1 cycle too early.
[image]

davideschiavone, 2020-03-18:
Dear @GTumbush , I force pushed another change that should solve both ecall and mret.
For what concern the changes you want to do on `retire_delay` for instance, they way to do it is to fork the cv32e40p repository https://github.com/openhwgroup/cv32e40p/

The fork will be on your GitHub page, you made the changes, push, and the create a Pull Request (as I am doing, I created the fork under my page davideschiavone, then I create PULL REQUESTs to the OpenHWGroup page, when they solve the issue it will be merged in the master repository)

GTumbush, 2020-03-19:
I did as you instructed above but was only able to add the addition of insn_regs_write because retire_delay is not in the cv32e40p master. The change is easy, replace wait(retire_delay); with @(retire_delay);  Pull request https://github.com/openhwgroup/cv32e40p/pull/269

GTumbush, 2020-04-22:
Hi Davide,
 With fix 302 I'm seeing the same issue with the mret instruction updating mstatus 1 cycle too late. The behavior is just like the original waveform in which there is 1 clock, retire, 1 clock, mstatus update. 
https://user-images.githubusercontent.com/54555114/76337501-104c5380-62bd-11ea-8577-207e7e2b92ec.png

I'm not seeing the mret instruction causing two clocks, mstatus update,  and then retire like in https://user-images.githubusercontent.com/54555114/76981667-888bc800-6900-11ea-9eca-19bb03344274.png.

davideschiavone, 2020-04-24:
Hi @GTumbush 

is this message older than this one (https://github.com/openhwgroup/cv32e40p/issues/302#issuecomment-618030768) ?

GTumbush, 2020-04-24:
Hi Davide,
  No, it is new. With the current code base I see this issue. Sorry for the confusion. Is there a better way for me to manage the issues?

davideschiavone, 2020-04-24:
No actually this issue is my fault. I should have merged back in the day :)
Now if you merge the branch fix302 with master it should solve all these issue as well.

I integrated back a lost commit :( 

Let me know so that I can merge everything back to master 
thanks for your contribution,
Davide

GTumbush, 2020-04-24:
I merged fix302 with the master and all issues are fixed. That would be great to get this all merged in with master so they are not lost.

davideschiavone, 2020-04-25:
OK cool! I create the PR and get the things merged.

thanks and sorry again for confusion

davideschiavone, 2020-04-26:
Fix and merged now: https://github.com/openhwgroup/cv32e40p/pull/313

closed 2020-04-26 by davideschiavone

GTumbush, 2020-05-11:
Below is the updated and correct behavior of the ecall instruction. Note that for the ecall instruction mstatus and mepc update on the rising edge of Clk immediately before the retire event on the falling edge of Clk.

[image]