openhwfoundation/cv32e40p #258: VCS fails to compile correctly

Imperas, 2020-03-13:
Hi All
I am seeing an error from VCS, it does not appear to handle the generate statement correctly, but in particular the expression evaluation for when (i==)

> Warning-[SIOB] Select index out of bounds
> verilog/openhwgroup/cv32e40p/rtl/riscv_pmp.sv, 204
> "pmp_addr_i[(0 - 1)]"
>   The select index is out of declared bounds : [15:0].
>   In module instance : RISCY_PMP.pmp_unit_i 
>   In module : riscv_pmp.

Has anybody successfully run through VCS ?

MikeOpenHWGroup, 2020-03-25:
As far as I am aware, there are currently no VCS active users in the OpenHW Group community.  If you are interested in using VCS, we are certainly interested in seeing it supported.

Please add a comment to this issue detailing your cwd and the exact command line used.

Also, you should be aware that the testbenches in this repository are not "official".    The official testbenches for cv32e40p are located in the OpenHW Group CORE-V verification repository, [core-v-verif](https://github.com/openhwgroup/core-v-verif).

Imperas, 2020-03-25:
We are trying to get VCS to work with the cv32e40p in ou step/compare
testbench and yes we are having issues - we are working with Synopsys on
this - and will be doing a shared screen with them shortly - we are close
to have things working - will keep you posted.

Simon
Imperas Software

-- 
====================================================================
The information contained in this electronic mail message and any
attachments hereto
is privileged and confidential information intended only for the use of the
individual or
entity named above or their designee.  If the reader of this message is not
the intended
recipient, you are hereby notified that any dissemination, distribution or
copying of this
communication is strictly prohibited.  If you have received this
communication in error
please immediately notify us by return  message or by telephone and delete
the
original message from your mail system.  Thank you.
====================================================================

MikeOpenHWGroup, 2021-01-03:
I'm reviewing old issues and just came across this one.  I will take the liberty of closing it as [core-v-verif](https://github.com/openhwgroup/core-v-verif) has supported VCS with CV32E40P for some months now.

closed 2021-01-03 by MikeOpenHWGroup